r-lidar/lidR #587: Error at lidR::delineate_crowns - Invalid number of points in LinearRing found

anayana, 2022-05-31:
see: https://stackoverflow.com/questions/72378483/how-to-solve-error-at-lidrdelineate-crowns/72380005#72380005

The code worked with old versions of R and lidR on my system, but doesn't with the new versions and functions.
running the code results in following error:

Error in (function (cond)  : 
                error in evaluating the argument 'x' in selecting a method for function 'addAttrToGeom': Evaluation error:
IllegalArgumentException: Invalid number of points in LinearRing found
3 - must be 0 or >= 4.

As the code seems to work on other systems, it doesn't at mine. What could be the problem?

Windows 10
R 4.1.1
lidR 4.0.1
all packages updated

Jean-Romain, 2022-05-31:
I confirm the bug is reproducible on Windows. But it does not come from lidR.

You should use more modern function based on `sf`

```r
tree.hulls <- crown_metrics(so_sample_las,  func = NULL,  geom = "concave", attribute = "treeID")
```

The error comes from the conversion to `sp` which works on Linux.

```r
sf::as_Spatial(tree.hulls.2 )
#> Error in (function (cond) :
#> erreur d'ï¿½valuation de l'argument 'x' lors de la sï¿½lection d'une mï¿½thode pour la fonction 'addAttrToGeom' : Evaluation #> error: IllegalArgumentException: Invalid number of points in LinearRing found 3 - must be 0 or >= 4
```

Please show me the output of 

```r
library(sf)
#> Linking to GEOS 3.10.1, GDAL 3.4.0, PROJ 8.2.0; sf_use_s2() is TRUE
```

I'll report a bug to `r-spatial/sf`

closed 2022-05-31 by Jean-Romain

anayana, 2022-06-01:
After updating packages (installing packages from github) processing the code results still in an error:

> library(sf)
Linking to GEOS 3.10.2, GDAL 3.4.1, PROJ 7.2.1; sf_use_s2() is TRUE
> library(lidR)
lidR 4.0.2 using 4 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/r-lidar/lidR>.

> so_sample_las <- readLAS("so_sample_las.laz")
> tree.hulls <- lidR::delineate_crowns(so_sample_las, type = "concave", attribute="treeID")
Warning messages:
1: 21 invalid polygons created. They likely correspond either to trees with aligned points or to edge cases where the convex hull converged to polygons that are not valid. 
2: 21 geometries were discarded before to convert from sf to sp. To keep them use crown_metrics() 

I tried it on a windows machine and a linux machine. Twice the same error.
A possible reason could be that the changes you have done are not in implemented yet in the packages I'm using. 
Is there a version of the package available where the changes you have done are implemented already?

Jean-Romain, 2022-06-01:
Did you read my answer above? Did you read the message printed? Everything work as expected and there is no error.

anayana, 2022-06-01:
SRY!!! I have read your answer and the message... However, I was inattentive. 
thx for your help! It's working!