r-lidar/lidR #595: problem when pulses have points with identical ReturnNumber

wiesehahn, 2022-06-14:
I get `Error: Invalid data: gpstime contains some NAs` when I try to read a certain LAS-file.

`las_check()` returns
`  - Checking gpstime incoherances
    ✗ 2719 pulses (points with the same gpstime) have points with identical ReturnNumber`

Is the error strict or is it possible to read the file and dismiss these pulses?

I expected the LAS-file to be read anyway, without these pulses but with a warning similar to `Warning messages: 1: Invalid data: 72 points with a return number equal to 0 found.` instead of an error.

In my case this is not a problem since `las <- readLAS(LASfile,select = "xyzc")` is enough and works fine, but is there a good way to fix these returns otherwise?

Jean-Romain, 2022-06-14:
It is not possible to store NA in gpstime. So it is impossible to read NA from a LAS file. It looks like a bug. Please share a reproducible example.

Error vs warning in `las_check()` are somewhat subjective.  Error are supposed to prevent some functions to work while warning are not supposed to. For example gpstime problems will prevent `track_sensor()` to work but some return number = 0 should not prevent any function to work. Yet, even if the functions work it does not mean the results are correct.

Your file contains invalid data for sure. But the reason is impossible to know, it is your responsibility to figure out why some return numbers are 0 for example. Maybe the data provider voluntarily included those points assuming you will filter them. Maybe it is an error from previous processing. Maybe the data provider made errors. Maybe two sources of data have been mixed. Maybe is is a small an fixable issue, maybe not. You alone can know what to do, this is why there is not tools to fix that automatically in `lidR`.

Jean-Romain, 2022-06-14:
Yes I need the file of course.

Jean-Romain, 2022-06-14:
I confirm there is a single point with `NaN` as gpstime. This is also the single point with a `PointSourceID` of 65382. Clearly this point is corrupted. Rewrite your file discarding PointSourceID = 65382 will solve the problem. Yet I do not understant

```r
library(lidR)
las = rlas::read.las("lasfilez_584000_5733000_laz.laz")
sum(is.na(las$gpstime))
i = which(is.na(las$gpstime))
las[(i-5):(i+5)]
#>           X       Y       Z  gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngleRank UserData PointSourceID
#> 1: 584238.5 5733563 192.576 577019.8       141            1               1                 0                0             20          FALSE         FALSE         FALSE            19       43           102
#> 2: 584238.1 5733563 192.582 577019.8       122            1               1                 0                0             20          FALSE         FALSE         FALSE            19       43           102
#> 3: 584237.7 5733563 192.581 577019.8       114            1               1                 0                0             20          FALSE         FALSE         FALSE            19       44           102
#> 4: 584237.2 5733563 192.566 577019.8       139            1               1                 0                0             20          FALSE         FALSE         FALSE            19       43           102
#> 5: 584236.8 5733563 192.570 577019.8       115            1               1                 0                0             20          FALSE         FALSE         FALSE            19       45           102
#> 6: 584236.4 5733563 192.548      NaN       135            1               1                 0                0             20          FALSE         FALSE         FALSE            19       44         65382
#> 7: 584200.4 5733568 194.654 577019.8        19            1               3                 0                0             20          FALSE         FALSE         FALSE            22       47           102
#> 8: 584200.1 5733568 193.830 577019.8        25            2               3                 0                0             20          FALSE         FALSE         FALSE            22       49           102
#> 9: 584199.4 5733568 192.198 577019.8        76            3               3                 0                0             20          FALSE         FALSE         FALSE            22       44           102
#>10: 584200.2 5733568 195.399 577019.8        21            1               3                 0                0             20          FALSE         FALSE         FALSE            22       53           102
#>11: 584199.7 5733568 194.152 577019.8        28            2               3                 0                0             20          FALSE         FALSE         FALSE            22       75           102
```

closed 2022-06-21 by Jean-Romain