r-lidar/lidR #596: Coordinate system error after upgrade

AndresKasekamp, 2022-06-16:
After upgrading from version 3.2 to 4.0.1, I am getting the following error when running `rasterize_canopy()` or `grid_canopy()`:

```
Error: no applicable method for 'st_crs<-' applied to an object of class "bbox"
In addition: Warning message:
In showSRID(uprojargs, format = "PROJ", multiline = "NO", prefer_proj = prefer_proj) :
  Discarded datum Unknown based on GRS80 ellipsoid in Proj4 definition
```

The problem was not there in earlier versions. Point cloud coordinate system is EPSG:3301.
I am getting an error with all my LAZ files. Here is one example: https://drive.google.com/drive/folders/1VIaOSldMLDMCg77yor64R6Koj5NJb0u3?usp=sharing

Jean-Romain, 2022-06-16:
I cannot read the file. It crashes if I try to read the extrabyte attributes. It works if I drop the extrabytes attributes

```r
las = lidR::readLAS("422384_2019_mets.laz", select = "* -0")
```

Can you confirm this behavior.

I have no issue with CRS. Please report a minimal reproducible example

```r
lidR::rasterize_canopy(las, 1, lidR::p2r())
```

AndresKasekamp, 2022-06-16:
When using the example above, I am still getting an error.
```
Error in UseMethod("st_crs<-") : 
  no applicable method for 'st_crs<-' applied to an object of class "bbox"
```
Could I be missing some underlying spatial libraries?

AndresKasekamp, 2022-06-16:
I managed to fix the issue by uninstalling sf and then reinstalling sf package.

Jean-Romain, 2022-06-16:
Good :+1: . But what about the crash at read time?

AndresKasekamp, 2022-06-17:
From my understanding, the extra attributes from that particular year were missing or erroneous which causes a lot of the programs to crash. The fix is to not read in the extra byte attributes.

Jean-Romain, 2022-06-17:
Ok so if other programs crash, I'll leave it as is

closed 2022-06-17 by Jean-Romain