r-lidar/lidR #600: merge_spatial only uses attribute value of last row in sf

jackbuehner, 2022-06-21:
I am attempting to merge an elevation attribute from buildings in a shapefile into the points that are within the bounds of the building. However, the value of the merged attribute does not correspond to the correct buildings; it only uses the value from the last building in the shapefile attribute table.

Below is the script I using:
```R
if(!require('lidR')) { install.packages('lidR') }
if(!require('rgdal')) { install.packages('rgdal') }
if(!require('sf')) { install.packages('sf') }

# LAS
lasFilePath = 'gvl_selection.las'
las = lidR::readLAS(lasFilePath)
lasCRS = sf::st_crs(las)

# buildings
buildingsSPDF = rgdal::readOGR(dsn = './', layer = 'gvl_selection_buildings', )
buildings = sf::st_transform(sf::st_as_sf(buildingsSPDF), lasCRS)

# spacial merge
lasWithBuildingMarkers = lidR::merge_spatial(las, buildings, 'TOPELEV') # expect

# expect different building elevations, but it
# uses the building elevation from the last building in the shapefile
print(lasWithBuildingMarkers$TOPELEV)
```

And here is a folder containing the las and shapefile: https://furman-my.sharepoint.com/:f:/g/personal/jbuehner_furman_edu/EohDIqayGWdGvt93eqku_LUBE3q2tiknRwVpHBRGmfBK7w?e=mBypDG

I have also attempted to merge without specifying an attribute name, and instead of applying the id number of each building (there are 14 buildings), it just applies either NA or 14.

Jean-Romain, 2022-06-21:
Your link points to an empty folder

jackbuehner, 2022-06-21:
Sorry, the upload didn't finish for some reason. Files should be there now

Jean-Romain, 2022-06-21:
The output looks good to me

```r
plot(lasWithBuildingMarkers, color = "TOPELEV")
```

[image]

jackbuehner, 2022-06-27:
I'm not sure what was going on, but it is working for me now too. Thank you for your help!

closed 2022-06-27 by jackbuehner