r-lidar/lidR #609: Error using `by_echo` for echo type "lastofmany"

candelas762, 2022-08-02:
When using `by_echo` in any of the "_metrics" functions the metrics calculated for the `lastofmany` echoes are `NA`. 

Here is an example using some functions:
```
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
las <- readLAS(LASfile, filter = "-keep_random_fraction 0.5")

echo = c("all", "first", "intermediate", "lastofmany", "single", "multiple")

# ================
# CLOUD METRICS
# ================

m = cloud_metrics(las, .stdmetrics_z, by_echo = echo)

col = names(m)[grepl("zmax", names(m))]
unlist(m[col])
> unlist(m[col])
             zmax        zmax.first zmax.intermediate   zmax.lastofmany       zmax.single     zmax.multiple 
            29.97             29.97             24.01                NA             29.97             28.56 

# ================
# PIXEL METRICS
# ================

m <- pixel_metrics(las, .stdmetrics_z, 20, by_echo = echo)

col = names(m)[grepl("zmax", names(m))]
unlist(m[col])
> unlist(m[col])
  class       : SpatRaster 
  dimensions  : 13, 12, 6  (nrow, ncol, nlyr)
  resolution  : 20, 20  (x, y)
  extent      : 684760, 685000, 5017760, 5018020  (xmin, xmax, ymin, ymax)
  coord. ref. : NAD83 / UTM zone 17N (EPSG:26917) 
  sources     : memory  
                memory  
                memory  
              ... and 3 more source(s)
  names       :  zmax, zmax.first, zmax.~diate, zmax.~fmany, zmax.single, zmax.~tiple 
  min values  :  0.00,       0.00,        5.36,         NaN,        0.00,       10.88 
  max values  : 29.97,      29.97,       24.01,         NaN,       29.97,       28.56 

# ================
# PLOT METRICS
# ================

shpfile <- system.file("extdata", "efi_plot.shp", package="lidR")
inventory <- sf::st_read(shpfile, quiet = TRUE)

m <- plot_metrics(las, .stdmetrics_z, inventory, radius = 11.28,by_echo = echo)

col = names(m)[grepl("zmax", names(m))]
summary(m[,col])
> summary(m[,col])
      zmax         zmax.first    zmax.intermediate zmax.lastofmany  zmax.single    zmax.multiple            geometry
 Min.   :21.61   Min.   :21.61   Min.   :14.92     Min.   : NA     Min.   :21.15   Min.   :21.61   POINT        :5  
 1st Qu.:25.26   1st Qu.:25.26   1st Qu.:18.26     1st Qu.: NA     1st Qu.:25.26   1st Qu.:23.25   epsg:26917   :0  
 Median :25.26   Median :25.26   Median :19.05     Median : NA     Median :25.26   Median :24.60   +proj=utm ...:0  
 Mean   :24.84   Mean   :24.84   Mean   :18.71     Mean   :NaN     Mean   :24.73   Mean   :24.13                    
 3rd Qu.:25.89   3rd Qu.:25.89   3rd Qu.:20.41     3rd Qu.: NA     3rd Qu.:25.89   3rd Qu.:25.03                    
 Max.   :26.18   Max.   :26.18   Max.   :20.89     Max.   : NA     Max.   :26.11   Max.   :26.18                    
                                                   NA's   :5                                                        

# ================
# CROWN METRICS
# ================

# Already tree-segmented point cloud
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
trees <- readLAS(LASfile, filter = "-drop_z_below 0")

metrics <- crown_metrics(trees, .stdtreemetrics, by_echo = echo)

col = names(metrics)[grepl("Z.", names(metrics))]
summary(metrics[,col]) # Here also "intermediate" echoes return NA metrics
> summary(metrics[,col]) # Here also "intermediate" echoes return NA metrics
    Z.first      Z.intermediate  Z.lastofmany    Z.single       Z.multiple             geometry  
 Min.   : 2.76   Min.   : NA    Min.   : NA   Min.   : 2.76   Min.   : 4.24   POINT Z      :200  
 1st Qu.:19.09   1st Qu.: NA    1st Qu.: NA   1st Qu.:18.80   1st Qu.:17.42   epsg:26912   :  0  
 Median :21.59   Median : NA    Median : NA   Median :21.55   Median :20.41   +proj=utm ...:  0  
 Mean   :21.03   Mean   :NaN    Mean   :NaN   Mean   :20.90   Mean   :19.86                      
 3rd Qu.:24.08   3rd Qu.: NA    3rd Qu.: NA   3rd Qu.:24.08   3rd Qu.:22.62                      
 Max.   :32.07   Max.   : NA    Max.   : NA   Max.   :32.07   Max.   :31.94                      
                 NA's   :200    NA's   :200                   NA's   :2                          
```

Jean-Romain, 2022-08-02:
I confirm the bug for `cloud_metrics`. All the functions are built on the same code base so  by inheritance they should all behave the same. The example can be made minimal with a constant metrics. We surprisingly get NAs as well meaning that there is something very wrong somewhere.

```r
m = cloud_metrics(las, .(m = 1), by_echo = echo)
unlist(m)
#>   m        m.first m.intermediate   m.lastofmany       m.single     m.multiple 
#>   1              1              1             NA              1              1 
```

Jean-Romain, 2022-08-02:
This should now be fixed

closed 2022-08-02 by Jean-Romain

candelas762, 2022-08-02:
How can I get the update? I have tried to update `lidR` and reinstall it but still having the error.

Jean-Romain, 2022-08-02:
You must install the version from github

candelas762, 2022-08-02:
Thanks. Now I confirm it is working also for me 👍

commit b350f4b8 (2022-08-02): Fix #609