r-lidar/lidR #610: Manual Tree Identification Suggestion

Saadi4469, 2022-08-03:
Good day,

Based on this [issue](https://stackoverflow.com/questions/73225889/r-rgl-lidr-slow-rendering-on-windows-11-64-bit) I would like to suggest an optional way to manually correct tree tops if possible by overlaying the points on an aerial basemap or a high resolution raster. This might increase the performance of the process. I am also open to other solutions as well.

Thank you

Jean-Romain, 2022-08-03:
You cannot dynamically edit a 2D plot so what is propose in lidR in 3D is not reproducible in 2D with regular plots

Jean-Romain, 2022-08-03:
Well actually it  is already working

```r
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las = readLAS(LASfile)
chm = rasterize_canopy(las)
ttops = locate_trees(chm, manual())
```

closed 2022-08-03 by Jean-Romain

Saadi4469, 2022-08-03:
Thank you, so is there a way fasten the rgl rendering for manual tree correction?

Jean-Romain, 2022-08-03:
If there was a way I know to get a faster rendering I swear I would already have put it in lidR :wink:

Saadi4469, 2022-08-03:
Oh cool thank you so much, I will give it a try tomorrow at work. However, I wish there was a way to fasten rgl 3D rendering for manual tree correction purposes.

Saadi4469, 2022-08-04:
Hope you had some fruitful discussion with the `rgl` maintainer.

Jean-Romain, 2022-08-04:
If you are referring to the person on stackoverflow  they are not the rgl maintainer. And we did not discuss at all actually. But I working on improving the `rgl` rendering for billion of points. I Think I have the ideas clear but the implementation is hard.

Saadi4469, 2022-08-04:
I am so happy to hear that, thank you so much for your time and help. I did post the same question [here](https://github.com/dmurdoch/rgl/issues/260), and the `rgl` maintainer did modify some of the code to improve performance. Good luck!

Saadi4469, 2022-08-04:
He did point out that the performance has to do with the CPU and not the GPU when it comes to rendering points. Moreover, `rgl` is single threaded and therefore does not use multiple thread, and the maintainer told me it's not possible to make `rgl` use multiple threads.