r-lidar/lidR #619: Individual Tree Segmentation

wiesehahn, 2022-09-07:
In the news for v4.0.1 it says that 

> Enhance: locate_trees() throws an informative error if called with an on-disk raster. The former error was cryptic. If the raster is small enough it is loaded on-the-fly.

now I guess I just ran into this problem: Trying to segment trees based on a CHM loaded with terra does not work, e.g.
```
chm <- terra::rast(imagepath)

f <- function(x) { x * 0.1 + 3}
ttops <- lidR::locate_trees(chm, lidR::lmf(f))

crowns <- lidR::silva2016(chm, ttops, max_cr_factor = 0.6, exclusion = 0.3, ID = "treeID")()
```
results in `Error: Cannot segment the trees from a raster stored on disk. Use segment_trees() or load the raster in memory`

But still I am not sure whats an efficient way to load the raster in memory. Maybe it would be nice to add this to the message.

Another confusion was that I expected the ITS algorithm to return tree crowns as sf vector data as is the case for treetop detection, but a raster was returned.

Jean-Romain, 2022-09-07:
Compared to `raster`, `terra` is somehow annoying with that because it never load a raster in memory and does not have a function to do it. The trick is to multiply by 1 to load in memory.

```r
chm = chm*1
```

> Another confusion was that I expected the ITS algorithm to return tree crowns as sf vector data

No, the algorithm is raster based. There is no reason to perform an additional vectorization step on the lidR side. This works that way since the very first version of lidR.

closed 2022-09-07 by Jean-Romain

wiesehahn, 2022-09-08:
thanks for your quick answers!