r-lidar/lidR #628: Output rasters have offset origin

zoeschindler, 2022-10-12:
Hello, I have an issue with the `rasterize_canopy()` function. The output raster is not centered at (0,0), this causes the raster to not perfectly overlap with some of my other rasters created in `lidR` and causes me further issues. The issue starts at resolutions lower than 0.5 meters, but I need a raster with 0.1 meters resolution. I noticed this also happens for example when creating rasters with `pixel_metrics()`.

Here is an example:

```
# load data
LASfile <- system.file("extdata", "Topography.laz", package="lidR")
las = readLAS(LASfile)

# calculate dsm
dsm <- rasterize_canopy(las, res = 0.5)
dsm@ptr$origin
> [1] 0 0

# calculate dsm
dsm <- rasterize_canopy(las, res = 0.1)
dsm@ptr$origin
> [1] 1.117587e-08 3.438443e-06
```

Jean-Romain, 2022-10-12:
Minimal reproducible example

```r
library(lidR)
LASfile <- system.file("extdata", "Topography.laz", package="lidR")
las = readLAS(LASfile)
u = lidR:::raster_layout(las, 0.1)
u = lidR:::raster_materialize(u)
u@ptr$origin
#> [1] 1.117587e-08 3.438443e-06
```

Jean-Romain, 2022-10-12:
More minimal reproducible example. The issue comes from `terra`

``` r
options(digits = 14)
b <- c(273357.1, 273642.9, 5274357.1, 5274642.9)
b <- terra::ext(b)
r <- terra::rast(b, resolution = 0.1)
r
#> class       : SpatRaster 
#> dimensions  : 2858, 2858, 1  (nrow, ncol, nlyr)
#> resolution  : 0.099999999999996, 0.099999999999935  (x, y)
#> extent      : 273357.1, 273642.9, 5274357.1, 5274642.9  (xmin, xmax, ymin, ymax)
#> coord. ref. :
r@ptr$origin
#> [1] 1.1117663234472e-08 3.4375116229057e-06
```

<sup>Created on 2022-10-12 by the [reprex package](https://reprex.tidyverse.org) (v2.0.1)</sup>

closed 2022-10-12 by Jean-Romain

zoeschindler, 2022-10-12:
Thank you very much for your efforts!

rhijmans, 2022-10-13:
@zoeschindler: there is not much that can be done about this imprecision. The "further issues" you mention can be addressed, as small differences like this are supposed to be ignored. But I do not know what the further issues are; so please report these, if you can.

zoeschindler, 2022-10-16:
@rhijmans I had issues when using `resample()`. I resampled to a higher resolution and got too many raster cells in my new raster, as the cells of the fine raster slightly overlapped the cells of the rough raster I resampled to. I mentioned it in https://github.com/rspatial/terra/issues/847. I am not sure how much of this problem is attributed to the slight offset or the issue with the function though.