r-lidar/lidR #637: use summary( ) or print( ) function report error

zzk112233, 2022-11-11:
Dear author：
When I use summary() or print () to check point cloud basic information， R always reports error ：
Error in if (is.na(units)) "units" else units : 
  argument is of length zero.
Can you tell me how to end this error?  Looking forward to your reply! Thank you~

Jean-Romain, 2022-11-11:
Please report a minimal reproducible example

zzk112233, 2022-11-12:
Here is example：
> library("lidR")
lidR 4.0.1 using 6 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/r-lidar/lidR>.
> las <- readLAS("G:\\LiDAR数据\\近自然经营LiDAR数据\\预处理数据\\裁切\\YW\\新建文件夹\\0.02 pre_ YW01CKK_2022_12.las")
> summary(las)
Error in if (is.na(units)) "units" else units : 
  argument is of length zero
> print(las)
Error in if (is.na(units)) "units" else units : 
  argument is of length zero

zzk112233, 2022-11-12:
Here is example：
```R
library("lidR")
lidR 4.0.1 using 6 threads. Help on <gis.stackexchange.com>. Bug report on <github.com/r-lidar/lidR>.
las <- readLAS("G:\LiDAR数据\近自然经营LiDAR数据\预处理数据\裁切\YW\新建文件夹\0.02 pre_ YW01CKK_2022_12.las")
summary(las)
Error in if (is.na(units)) "units" else units :
argument is of length zero
print(las)
Error in if (is.na(units)) "units" else units :
argument is of length zero
```

Jean-Romain, 2022-11-12:
Can you share the file or a subset of the file that allow to reproduce.

zzk112233, 2022-11-13:
Here is subset of the file:
[https://drive.google.com/file/d/1_3IBiQ98T-3e3o_uIcr4i34sbHMmCLjY/view?usp=sharing](url)

zzk112233, 2022-11-14:
Here is subset of the file
[https://drive.google.com/file/d/1_3IBiQ98T-3e3o_uIcr4i34sbHMmCLjY/view?usp=share_link](url)

Jean-Romain, 2022-11-14:
Fixed but your CRS seems invalid

commit d9c4aac2 (2022-11-14): Fix #637
closed 2022-11-14 by Jean-Romain