r-lidar/lidR #650: Please remove dependencies on **rgdal**, **rgeos**, and/or **maptools**

rsbivand, 2022-12-15:
This package depends on (depends, imports or suggests) **raster** and one or more of the retiring packages **rgdal**, **rgeos** or **maptools** (https://r-spatial.org/r/2022/04/12/evolution.html, https://r-spatial.org/r/2022/12/14/evolution2.html). Since **raster** `3.6.3`, all use of external FOSS library functionality has been transferred to **terra**, making the retiring packages very likely redundant. It would help greatly if you could remove dependencies on the retiring packages as soon as possible.

Jean-Romain, 2022-12-15:
`rgdal` is now removed.

However I don't know when I will be able to release on  CRAN. I can no longer release `lidR` on CRAN for months because it takes too long to check on Windows and the submission is systematically rejected.

For an unknow reason each step of `R CMD check` on CRAN's Windows machine takes 2 to 7 times longer than on Debian. The timing that was previously ok is no longer accepted and I'm tired of fighting against the constantly changing CRAN policies. I considered removing unit tests but this won't be enough to gain 2+ minutes. So, I eventually resigned and stopped updating the package because at this stage I must remove the unit tests and the documentation to get a chance to reach the target timing.

Jean-Romain, 2022-12-15:
Problem solved. For the first time for months Uwe Ligges triggered reverse depency despite timing problem. `lidR` is on its way.

closed 2022-12-15 by Jean-Romain