r-lidar/lidR #659: Error reading data with non-integer spectral attributes

seanreilly66, 2023-01-24:
readLAS fails to load datasets that contain non-integer values for default spectral attributes. Returns the following error message:

> Error: Invalid data: (Attribute name) is not an integer

Example:

```r
library(lidR)

las <- LAS(data = data.frame(
  X = runif(100, 1, 10),
  Y = runif(100, 1, 10),
  Z = runif(100, 1, 10)
))

las <- LAS(df)

nir = runif(100)

las <- add_lasattribute(las, x = nir, name = 'nir', desc = 'nir')

writeLAS(las, 'temp/nir_error.las')

readLAS('temp/nir_error.las')
```

Discussed in https://github.com/r-lidar/lidR/discussions/658

Jean-Romain, 2023-01-24:
Fixed thanks

Jean-Romain, 2023-01-24:
I really enjoin you to use `add_lasattribute_manual` to store your `nir` as float intead of `double`

commit 1f62510e (2023-01-24): Fix #659
closed 2023-01-24 by Jean-Romain