r-lidar/lidR #662: Watershed

felipenorconsult, 2023-02-03:
Hi,
 
I've been using your R package for tree segmentation quite some time, but suddenly it wouldn’t work for any reason:
 
> algo <- watershed(chm_s, th_tree = 4)
Error in watershed(chm_s, th_tree = 4) : unused argument (th = 4)

Jean-Romain, 2023-02-03:
Weird. I don't think it comes from lidR. Do you have another package loaded that provides a function named `watershed`? What if you write `lidR::watershed(chm_s, th_tree = 4)`

felipenorconsult, 2023-02-03:
Hi JR

Thanks a lot, that did the trick. 😊

Felipe Verdú
Teknikledare GIS och fjärranalys
Dir: +46 10 14 17 591 | Mob: +46 723 89 24 53
***@***.***
***@***.***
Norconsult AB
Box 8774, 402 76 Göteborg
Theres Svenssons gata 11, 417 55 Göteborg
Tel: +46 10 141 80 00
www.norconsult.se<http://www.norconsult.se/>
Every day we improve everyday life

CAUTION: External email

CONFIDENTIALITY AND DISCLAIMER NOTICE: This message is for the sole use of the intended recipients and may contain confidential information. If you are not an intended recipient, you are requested to notify the sender by reply e-mail and destroy all copies of the original message. Any unauthorized review, use, disclosure or distribution is prohibited. While the sender has taken reasonable precautions to minimize the risk of viruses, we cannot warrant the absence of, or accept liability for, any such viruses in this message or any attachment.

closed 2023-02-03 by Jean-Romain