r-lidar/lidR #663: Item 2 of j is 4 which is outside the column number range [1,ncol=2]

flottsam, 2023-02-10:
I'm trying to segment a las catalog and not sure how to interpret the error message: 
```
error occurred when processing the chunk 6. Try to load this chunk with:
 chunk <- readRDS(...) 
las <- readLAS(chunk)
Item 2 of j is 4 which is outside the column number range [1,ncol=2]
```
This particular tile has powerlines in it, which I suspect may be the issue. 

To find the trees, I used: 
` (trees <- locate_trees(ctg, algorithm = lmfauto(), uniqueness = "bitmerge"))`

And to segment: 
```
dir.create(here("als/4_segmented"))
opt_filter(ctg) <- "-drop_z_below 0 -drop_class 2"
opt_output_files(ctg) <- here("als/4_segmented/{ORIGINALFILENAME}")
opt_laz_compression(ctg) <- FALSE # output .laz or .las?
plan(multisession, workers = detectCores(logical = FALSE)/2) # use 1/2 physical cores
# run function 
silva <- silva2016(chm=chm, treetops = trees)
segment_trees(ctg, algorithm = silva, attribute = "treeID", uniqueness = "bitmerge")
```

Sometimes the error message is returned on a different tile. They are always (so far) tiles on the edge of the catalog.  

[image]

Jean-Romain, 2023-02-10:
I don't know what is the error. You must provide a reproductible exemple. When failing the LAScatalog processing engine save the chunk that failed. You can read it and run the function on this particular chunk to reproduce the exact condition that lead to a crash.

```r
chunk <- readRDS(...) 
las <- readLAS(chunk)
silva <- silva2016(chm=chm, treetops = trees)
segment_trees(las, algorithm = silva, attribute = "treeID", uniqueness = "bitmerge")
```

If the error is reproductible, you must send me some data so I can reproduce.

Thanks

flottsam, 2023-02-13:
OK here is a link to a subset of 6 tiles and some basic code for segmenting the point cloud:
https://drive.google.com/drive/folders/1O77tDnROkrs8VcQ3ABVInL8eRgbFKcEe?usp=sharing
Sometimes the error happens on a different tile. 
Let me know if have any issues accessing the data.
(It might take a while to upload the tiles, internet is very slow right now.)

Thank you.

flottsam, 2023-02-14:
I was able to get most of the way through processing by using a larger buffer size (`opt_chunk_buffer(ctg) <- 90`), but it failed again with `extents do not overlap` error message and warning messages: `In max(Z) : no non-missing arguments to max; returning -Inf `

[image]
[image]

flottsam, 2023-02-14:
So I just realized that I had cropped the CHM to a smaller boundary file, which obviously explains the non-overlapping extent error. I was able to finish processing the CTG by sub-setting the CTG to the CHM extent (sorry about that) and using a 90m buffer (the error was not clear to me if that was indeed the case). I'm curious about the `no tree found. Maybe use different parameters.` warning. Does that warning apply to the entire tile, or only if trees exist in the treetops (shapefile), but not in the CHM, like with powerlines, for example? 
[image]

flottsam, 2023-02-15:
There seems to be an issue with `uniqueness = "bitmerge"`. When I run `segment_trees(las=ctg_subset, algorithm = silva)`, tree segmentation runs to completition, as shown above. However, `segment_trees(las=ctg_subset, algorithm = silva, uniqueness = "bitmerge")` yields an error: `Item 2 of j is 4 which is outside the column number range [1,ncol=2]` and a message: `In max(Z) : no non-missing arguments to max; returning -Inf`

I have subset the the catalog to the CHM, but I noticed that one tile (top right) does not actually intersect spatially with the CHM, as shown below. The CHM, trees, and tiles all have the same CRS. 

[image]

flottsam, 2023-02-17:
Sorry to bombard you with so many replies, but the problem appears to be related to the tiles that do not spatially intersect w/ the CHM. Below better visualizes the problem. 
I first use `locate_trees(chm, lmf(f))`. Then, I subset the CTG with `ctg_subset <- catalog_select(ctg, chm)`, and then `ctg_trees <- segment_trees(ctg_subset, silva)`, and get the results below. Orange tiles do not have trees because the CHM does not exist there (_Warning: _1: No tree can be used as seed__ ). 

[image]

The problem is when I try to delineate crowns from the newly segmented CTG.  `crowns <- crown_metrics(ctg_trees, func = .stdtreemetrics, geom = "concave")`, because there are no trees in those tiles that do intersect spatially with the CHM. 
[image]

Jean-Romain, 2023-02-17:
I appreciate that you are trying to provide many details. However as you said, you bombarded me we data. At this stage I'm no longer able to sort out all the data you gave me.

Can you produce a clean and tidy minimal reproducible example using a small data file ? If the problem comes from seeds, chm and tiles that do not match I guess it should be easy to reproduce with the data shipped with the package.

flottsam, 2023-02-17:
I subset the ctg using a shapefile rather than the chm and was able to process w/o error. 
[image]

closed 2023-02-17 by flottsam