r-lidar/lidR #670: Is there the type of c++ about algorithm-its.R？

lizlin7, 2023-03-31:
Hello,author!I am very interested in the tree segmentation,but I do not know R very well.Could you tell me where is the type of c++  about the algorithm-its.R?I can not find it,thank you very much!

Jean-Romain, 2023-03-31:
Sorry I don't underground the question. Btw, please use the discussion tab to ask questions, This is the section to report bugs.

lizlin7, 2023-03-31:
I mean,as to the algorithm-its.R in R,whether there is a c++ version,I can not find it.Btw,I am sorry for asking questions here as I am not familiar with gitub now,I will pay attention to it in the furture.

Jean-Romain, 2023-03-31:
C++ is in the src folder https://github.com/r-lidar/lidR/tree/master/src

closed 2023-03-31 by Jean-Romain