r-lidar/lidR #671: ScanAngleRank

spono, 2023-04-08:
Ciao JR,
I'm having some issues with some new files I received.

Running an analysis I got this error:

```
> ads = clip_roi(ctg, shp, radius=fixed.radius)
> metrics = catalog_apply(ads, cloud_metrics, func = .my_metrics)
Error: object 'ScanAngleRank' not found
```
`.my_metrics` calls `ScanAngleRank` in a pretty regular way:

```
.my_metrics = ~my_metrics(X, Y, Z, ReturnNumber, Classification, ScanAngleRank,
                                        ht_slice=1, ht_max=NULL, min_pts=0, sar_thr=NULL)
```

I checked the data in R and it seems that `ScanAngleRank` is reported as `ScanAngle`.

```
> head( las@data )
          X       Y     Z   gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification
1: 685999.3 5089999  0.27 336380160        86            1               1                 0                0              1
2: 685999.4 5089987 -0.24 336380160        16            2               2                 1                0              1
3: 685999.3 5089989 -0.25 336380160        52            2               2                 1                0              1
4: 685999.6 5089999  9.50 336380160         8            3               4                 1                0              1
5: 685999.0 5089997  0.46 336380160         7            4               4                 1                0              1
6: 685998.9 5089998  0.18 336380160        78            1               1                 1                0              1
   ScannerChannel Synthetic_flag Keypoint_flag Withheld_flag Overlap_flag ScanAngle UserData PointSourceID
1:              0          FALSE         FALSE         FALSE        FALSE   -13.998        0          1253
2:              0          FALSE         FALSE         FALSE        FALSE   -13.998        0          1253
3:              0          FALSE         FALSE         FALSE        FALSE   -13.998        0          1253
4:              0          FALSE         FALSE         FALSE        FALSE   -13.998        0          1253
5:              0          FALSE         FALSE         FALSE        FALSE   -13.998        0          1253
6:              0          FALSE         FALSE         FALSE        FALSE   -13.998        0          1253
```

As a crosscheck, I tried to load a LAZ in CloudCompare and the weird thing is that `ScanAngleRank` is there.

[image]

Looking at the header, there's nothing I can imagine could provide an insight.

```
> las@header
File signature:           LASF 
File source ID:           0 
Global encoding:
 - GPS Time Type: Standard GPS Time 
 - Synthetic Return Numbers: no 
 - Well Know Text: CRS is GeoTIFF 
 - Aggregate Model: false 
Project ID - GUID:        00000000-0000-0000-0000-000000000000 
Version:                  1.4
System identifier:         
Generating software:      rlas R package 
File creation d/y:        284/2022
header size:              375 
Offset to point data:     375 
Num. var. length record:  0 
Point data format:        6 
Point data record length: 30 
Num. of point records:    4986157 
Num. of points by return: 3480656 1168058 220348 117095 0 0 0 0 0 0 0 0 0 0 0 
Scale factor X Y Z:       0.01 0.01 0.01 
Offset X Y Z:             0 0 0 
min X Y Z:                683999.9 5088000 -22.16 
max X Y Z:                686000.1 5090000 42.49 
Variable Length Records (VLR):  void
Extended Variable Length Records (EVLR):  void
```

Any hint? let me know if you need a file

Jean-Romain, 2023-04-08:
> I then modified the function in order to work with ScanAngle and, in the end, I got this:

What did you do to get this error ?

In the las specification `ScanAngleRank` is an integer attribute which is supposed to record the rank of the scan angle. 0 is not 0 degree but the 0th angle of the sampling device. Usually we are assuming that the value `n` corresponds to `n` degrees and it is probably what is recorded. The `ScanAngle` on another hand records the actual off nadir angle  with decimal numbers (after conversion from raw data). So  scan angle and scan angle rank are two different things and in `lidR` I choose to make them different. Some software may opt to call them the same but in `rlas` I opt to stay as close as possible to the spec.

So basically you juste have to call your metrics function

```r
mymetrics(X,Y,Z, ScanAngle)
```

instead of

```
mymetrics(X,Y,Z, ScanAngleRank)
```

spono, 2023-04-11:
> mymetrics(X,Y,Z, ScanAngle)

great, that's what I did in the meatime as a workaround.

> What did you do to get this error ?

(I'm out of the office, I'll let you know as soon as I replicate it)

closed 2023-04-12 by Jean-Romain