r-lidar/lidR #675: Filter options in readLAScatalog are superseded by opt_filter().

rs806, 2023-04-26:
Can the filter in readLAScatalog be made to be persistent? For example, I apply a filter when reading in a catalog, and then I am lazy and don't repeat that filter after adding a new one with opt_filter(), and then I lose the first filter.

```
library(sf)
library(lidR)

smallCat <- readLAScatalog(system.file("extdata", "example.laz", package = "rlas"),
  filter = "-drop_z_below 975")
plotDF <- data.frame(place = "plot_1", x = 339008, y = 5248001)
sf_SAMPLE <- st_as_sf(x = plotDF, coords = c("x", "y"), crs = st_crs(smallCat))

plotRadius <- 11.35

myMetrics = function(z)
{
  res = list(
    zmin = min(z),
    zmax = max(z),
    zmean = mean(z)
  )
  return(res)
}

# -------------------------------------------------------------------------------------------------
# problem happens here.
# This line will supersede filter set in readLAScatalog.
# i.e., if it is not run, minimum value is 973.145. If it is run, minimum value is 975.182
# -------------------------------------------------------------------------------------------------
opt_filter(smallCat) <- "-drop_withheld"
# -------------------------------------------------------------------------------------------------

plotTreeVolAndType <- plot_metrics(smallCat,func = ~myMetrics(Z), sf_SAMPLE, radius = plotRadius)
print(plotTreeVolAndType$zmin)
print(plotTreeVolAndType$zmax)

```

Jean-Romain, 2023-04-26:
In summary you want a new function `opt_add_filter()`. I'm I correct ?

rs806, 2023-04-26:
Initially I thought it would be better to always maintain the filter set up when running readLAScatalog(). In my actual project I drop withheld right away knowing that I'll never want to include those. However I can see your suggestion of opt_add_filter() will provide flexibility in other cases. But now that I'm thinking about this, maybe it would just be better to warn the user that the previous filters will be lost when using opt_filter(). This would remind me to always explicitly choose filter options. I see either of those options. Thanks.

Jean-Romain, 2023-04-27:
For backward compatibility I cannot change the current behavior anyway. I can only add a new convenient function

rs806, 2023-04-27:
I see now from the LAS catalog vignette that readLAS() behavior is different in that it does discard filtered hits. So I needed to be more careful with readLAScatalog(). For that reason it's probably better for me to just state all the filters I want each time I use opt_filter(). My feeling is that adding a new function opt_add_filter() wouldn't be necessary. Thanks for taking the time to look at this!

closed 2023-04-27 by rs806