r-lidar/lidR #680: plot extent of las file with mapview

wiesehahn, 2023-05-10:
`lidR::plot()` is able to plot a LAScatalog with mapview (e.g. `plot(ctg, mapview=TRUE)`). 

With single LAS files this it not possible. Is there a reason for this or would it be possible to implement it so that  `plot(las, mapview=TRUE)` will plot the extent of the LAS file on top of some basemap with mapview?

wiesehahn, 2023-05-10:
thanks, exactly that. 
Just seems a little inconsistent, but now I know.

Jean-Romain, 2023-05-10:
`plot(las)` plots in 3D this is the default behavior expected by users I guess. I could add something like `plot(las, 2d = TRUE)` but this would also be inconsistant. I'm open to suggestion if you have an idea.

wiesehahn, 2023-05-10:
I am fine with the current state now that I know how. 
But intuitively I expected that `plot(x)` returns a 3D plot (default and most expected by users) for LAS files and plot(x, mapview=TRUE) returns a 2D plot similar to the case if x is a LAScatalog.

Jean-Romain, 2023-05-10:
This argument does not exist for `plot.LAS`. It only exists for `plot.LAScatalog` but I could add it. That makes sense.

Jean-Romain, 2023-08-16:
It is now possible to use `plot(las, mapview = TRUE)`

commit 745974c9 (2023-08-16): fix #680
closed 2023-08-16 by Jean-Romain