r-lidar/lidR #684: Error in data.table::setnames(where, c("X", "Y")) : Can't assign 2 names to a 1 column data.table

FloFranz, 2023-05-16:
I have a list of laz files thinned before (name: `thinned_pc_list`) and want to rasterize them with the point-to-raster algorithm. For one element of the laz files list, I get the error message: `Error in data.table::setnames(where, c("X", "Y")) : Can't assign 2 names to a 1 column data.table`. When increasing the subcircle radius of the disks, the error does not appear. Also when leaving the `na.fill` argument, it works. So I think it's related to the lower number of NA pixels resulting from a larger subcircle radius. Here's my code:

`dsm_test <- lidR::rasterize_canopy(thinned_pc_list[[3]], res = 1, algorithm = lidR::p2r(subcircle = 0.25, na.fill = lidR::tin()))` --> that works

`dsm_test <- lidR::rasterize_canopy(thinned_pc_list[[3]], res = 1, algorithm = lidR::p2r(subcircle = 0.2, na.fill = lidR::tin()))` --> here I get the error

Why does it appear only for this one laz file and do you know a way to fix it?

FloFranz, 2023-05-16:
You can find it in this [repository](https://github.com/FloFranz/cloud_2_dsm). It happens in script `cloud2dsm` inside a loop (line 272-276), but I added one line of code (line 279) where you can find exactly the code above to test it. All what you need are some laz files in the folder `data\raw_data\dsm_cloud` and corresponding laz files containg ground points in `data\raw_data\dtm_tiles`. But I think it would be best if you had my files, since the problem is only with one particular laz file. Unfortunately, they are to large to push them, even when I want to push two of them. Is there another possiblity to provide you with the files (one where it works and the one where it doesn't)? Otherwise I will work on to provide you a minimal reproducible example maybe later.

Jean-Romain, 2023-05-16:
Please give me a minimal reproducible example. I'm not going to dig into 400 lines of third party code.

FloFranz, 2023-05-17:
```
# read several las files into a list
filepath <- path/to/folder/with/lazfiles   # set filepath to the las files
las_files <- list.files(filepath, full.names = TRUE)

las_files_list <- c()
for (i in seq_along(las_files)) {
  
  las_file <- las_files[i]
  las <- lidR::readLAS(las_file)
  las_files_list[[i]] <- las
  
}

# calculate DSM rasters from the las files
dsm_list <- lapply(las_files_list,
                   FUN = function(x)
                     lidR::rasterize_canopy(x,
                                            res = 1,
                                            algorithm = lidR::p2r(subcircle = 0.2,
                                                                  na.fill = lidR::tin())))
```

I provide you two example laz files [here](https://nextcloud.nw-fva.de/index.php/s/YoQca7Df5TmbK87) (one where it works and the one where it doesn't). I've sent you the password by e-mail.

Jean-Romain, 2023-05-25:
Fixed thanks

commit b4f9966b (2023-05-25): Fix #684
closed 2023-05-25 by Jean-Romain