r-lidar/lidR #692: Read .copc.laz files

ouroukhai, 2023-07-24:
Hello,
is there a way to read .copc.laz files directly from lidR? The only solution I've found so far is to convert to .laz via PDAL library.
Thank you for the time

closed 2023-07-24 by Jean-Romain

wiesehahn, 2023-07-24:
That sounds great, does this also allow a more performant plotting experience? E.g. take advantage of the internal octree tiling to render only points in sight.

Jean-Romain, 2023-07-24:
> take advantage of the internal octree tiling to render only points in sight.

No. But I will probably work on it later. I have some existing code that may be used for that. Yet doing that with `rgl` is not straightforward at all.

Jean-Romain, 2023-11-23:
The github version of rlas has COPC support. https://github.com/r-lidar/rlas. On CRAN soon.