r-lidar/lidR #694: error in segment_trees()

Jean-Romain, 2023-07-24:
### Discussed in https://github.com/r-lidar/lidR/discussions/693

<div type='discussions-op-text'>

<sup>Originally posted by **taeyoonlee87** July 24, 2023</sup>
Hello, 

I tried to segment individual trees. 

I created CHM using 'rasterize_canopy' and created the tree tops using 'locate_trees'. 

I applied "st_as_sf" to convert tree tops to SF objects. Like below 

```r
opt_output_files(las_cat) <- ""
ttops <- locate_trees(las_cat, lmf(5), uniqueness = "bitmerge")
ttops2<- st_as_sf(ttops)
```

Then, I segment it using "segement_trees"

```r
opt_output_files(las_cat) <- "/scratch/tl96575/mapping/chm/segmented_{XLEFT}_{YBOTTOM}"
algo <- silva2016(chm, ttops2)
ctg_segmented <- segment_trees(las_cat, algo)
```

but it resulted in the error message 

```r
#Chunk 1 of 635 (0.2%): state ✗
#Error: On-disk rasters not supported in 'raster_as_dataframe()'
```

When I test my code with a small portion of data, it worked without errors. But when I ran my code for the whole dataset, it gave me this error. 

Did you ever see this error? Or could you please advise me on what point I need to check.?</div>

taeyoonlee87, 2023-08-15:
want to update you.  
When I run my code for the whole dataset (625 las files), it gave me an error 
```
> crown_metrics(ctg_segmented, NULL, geom = "convex")
Error in lazyeval::f_capture(func) : Promise has already been forced
Calls: crown_metrics -> crown_metrics.LAScatalog -> <Anonymous>
```
When I ran my code for partial data (20~30 las files), it gave the results that I wanted. 

However, I also found that some las files were missing. 

Do you think does it relate to the bug? 

Bests,

Jean-Romain, 2023-08-15:
Not related at all for sure. Open a new issue with a reproducible exemple

Jean-Romain, 2023-08-16:
MRE

```r
library(lidR)
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
las <- readLAS(LASfile)
col <- pastel.colors(200)

chm <- rasterize_canopy(las, res = 0.5, p2r(0.3))
f = tempfile(fileext = ".tif")
terra::writeRaster(chm, f)
chm = terra::rast(f)

ttops <- locate_trees(chm, lmf(4, 2))
las   <- segment_trees(las, silva2016(chm, ttops))
```

Jean-Romain, 2023-08-16:
Fixed

closed 2023-08-16 by Jean-Romain

taeyoonlee87, 2023-08-16:
Thanks !! 

Do I need to update the package to apply it?

Jean-Romain, 2023-08-16:
You need to install version 4.0.4 from github. It has not been release on CRAN yet

commit 018d9f90 (2023-08-16): Fix #694