r-lidar/lidR #713: Individual tree segmentation -> dalponte2016

utzdrechsler, 2023-09-21:
This is the code from the lidrBook that I have used.

algo <- dalponte2016(chm_p2r_05_smoothed, ttops_chm_p2r_05_smoothed)
algo <- segment_trees(las, algo) # segment point cloud
plot(las_algo, bg = "white", size = 4, color = "treeID") # visualize trees

The derived chm and ttop segmentation are from the chapter before and both work.
When I try to run dalponte2016 I get the followin error message:
Error in loadNamespace(x) : there is no package ‘future’

Anybody else got the same issue and knows what to do?

utzdrechsler, 2023-09-21:
Sorry it was very simple:

install.packages("future")
library(future)

Jean-Romain, 2023-09-21:
Simple to solve yet there is a bug. It should work without future.

Jean-Romain, 2023-12-08:
Fixed

commit 6b63774f (2023-12-08): Fix #713
closed 2023-12-08 by Jean-Romain