r-lidar/lidR #715: pixel_metrics() runs single threaded with plan(multisession, workers > 1) and readLAS()

twest820, 2023-09-24:
From #523 it looks like `pixel_metrics()` in lidR 4.0.4 should support parallel execution. However, I'm only seeing a single core go active.
```
tile = readLAS(filePath)
plan(multisession, workers = 16)

metrics = pixel_metrics(tile, .stdmetrics)
```
I'm guessing this is because #523 enabled parallelism at the inter-tile (catalog) level but not within individual tiles. It'd be nice to also have lower level parallelism.

For example, at the moment I've 650 some tiles to process with a metrics function which takes about 20 thread-minutes and 20–30 GB DDR per tile. Given 128 GB of DDR, four or five jobs can thus load tiles in parallel without exhausting physical memory to the point of excessive swapping. Given a 16 core (all hyperthreaded) processor, each job can be allocated 4–8 threads, resulting in nominal runtime of something like 20 minutes/tile * 650 tiles / (5 jobs * 6 threads/job) = 7.2 hours. Since `pixel_metrics()` currently runs single threaded in this use case, the implementable divisor is 1 thread/job and the run time goes up close to two days.

A partial workaround's presumably to build a catalog for each job but, since the catalog level metrics need to be sharded back into rasters corresponding to the original tiles, it'd be simpler if `pixel_metrics()` could just call the metrics function in parallel over a given tile. I'm unsure how easy R makes this to implement in practice but didn't come up with a search hit for an issue tracking intra-tile parallelism.

Jean-Romain, 2023-09-25:
parallelism within individual point cloud is not straightforward:

- The algorithm may not be parallelisable at all.
- The algorithm may call code from third party package that is not parallelized
- Some algorithm could be parallelisable but we must recall that R in single threaded and some limitations apply

What can be parallelized is already parallelized in lidR. In the specific case of `pixel_metrics` the function relies on `data.table`. `data.table` is already world the fastest and already makes use of parallelism on key locations. There is nothing I can do to speed up `pixel_metrics`. At least not in lidR.

twest820, 2023-09-25:
> The algorithm may not be parallelisable at all.

Not the case for `pixel_metrics()`, though, since it's a for loop over the cells in a raster. Also not the case for `.stdmetrics` and such as the calculations for each cell are independent of all others and thus parallelize trivially.

> The algorithm may call code from third party package that is not parallelized

Sure, but 1) that's not the case for `pixel_metrics()` itself and 2) it's hard to see why someone somewhere potentially passing a thread unsafe function to `pixel_metrics()` sometime should force everyone else's thread safe code to run more slowly.

>  we must recall that R in single threaded and some limitations apply

Which is presumably why lidR already composes with future and uses OpenMP.

> There is nothing I can do to speed up pixel_metrics. 

Calling a function pointer within a parallel for loop's not difficult from C++. Since lidR already implements such loops and has C++ metrics functions it's unclear this statement holds, particularly since stdmetrics kinds of operations are pretty stable and could also be moved into C++.

The part I'm not entirely sure about is calling back into user provided R functions.. Since calling R from C++ is supported there's no fundamental obstruction to most such functions but, as you've pointed out, R is sometimes lacking routine basic thread safety. So it's possible internal R issues might prevent use of thread safe user functions. In which case an R core issue can be opened for fix.

R core would probably want a repex for that, though, and `pixel_metrics()` current state prevents attempting trying to write one.

Jean-Romain, 2023-09-25:
In theory you are correct. In practice you are wrong. All `*_metrics`  function are based on `data.table` aggregation which means they all run through a single main function in lidR. Internally they rely on a super efficient and safe code from `data.table`. There is one exception however for `point_metrics` that uses parallelized code and pointers to R functions exactly the way you are describing. And trust me it is complex and it partially relies on undocumented parts of the poorly documented R's C API. However my code is not as safe as the `data.table` code because nothing replace decades of development by world class programmers. 

I don't want to write custom C++ code with call to R function pointers for every `_metrics`  functions. And even if I'd try it, it is not guaranteed I could beat `data.table`. And even if I beat `data.table` I'll run into so much potential troubles on CRAN.

Now if you thing you know R, lidR and the R's C API better than me (which is possible, I do not doubt about that), feel free to make a pull request with a parallelized version of `pixel_metrics()`.

twest820, 2023-09-25:
> feel free to make a pull request with a parallelized version of `pixel_metrics()`

That would be some years out, I'm afraid. Basically we'd need a letter of support from you for a grant application and sufficient off GitHub discussion as to have reasonable assurances a fundable range of PRs would be accepted. Prerequisite to that's probably some discussion with R core, which I'm unfortunately not going to have cycles to drive anytime soon.

If I can find the time I'll try to throw together a point solution and put it under the GPL.

closed 2023-09-25 by Jean-Romain

Jean-Romain, 2023-09-25:
Hey, honestly if you are not happy with lidR you can just stop using it. I won't take it personally.

Jean-Romain, 2023-10-20:
Implementation of what? A function equivalent to `pixel_metric()` capable of mapping any user defined function safely? Which user-defined function did you use? Did you include the time to read the laz files? What are you benchmarking? How many points? How many files? I don't even know what we are talking about.  What are you comparing?

A 6000x speed-up seems very unlikely. You would have come with a 10-fold speed-up, I could have taken it seriously.

By the way, you may be interested to know that I'm working on a new package. This package will be much more efficient than `lidR` yet it will be less versatile. My benchmarks for 2 laz files, 30 millions points, code in C with R's C API, no parallelization, no `data.table`, no `Rcpp` are:

- `pixel_metrics(ctg, 1, ~mean(Z))` : 50 seconds
- new project for strictly equivalent code: 40 seconds

90% of the time is actually reading the 2 laz files. The amount of time dedicated to applying `mean(Z)` per pixels is likely to be under a second.

So please do not put figures out of nowhere. If you claim you have done something extraordinary, at least show some reliable benchmarks.