r-lidar/lidR #716: No plot with plot function (and with plot_dtm3d)

Redcozmo, 2023-10-02:
Hello,

I'm testing lidR package. I've an issue with plotting simple plots with plot function.

With this code :

`LASfile <- system.file("extdata", "Topography.laz", package="lidR")`
`las <- readLAS(LASfile, select = "xyzc")`
`plot(las, size = 3, bg = "white")`

Nothing is plotted.

But with backend='lidRviewer'

`plot(las, backend='lidRviewer', size = 3, bg = "white")`

The Point Cloud Viewer window appears (but without background require by my parameter)

I have the same issue with plot_dtm3d function

With this code :

`dtm_tin <- rasterize_terrain(las, res = 1, algorithm = tin())`
`plot_dtm3d(dtm_tin, bg = "white")`

Nothing is plotted.

Thanks for helping.

Jean-Romain, 2023-10-02:
Does `rgl` works? Try plotting a basic `rgl` graph without involving `lidR`

Redcozmo, 2023-10-02:
I think because this work : 

`dtm_tin <- rasterize_terrain(las, res = 1, algorithm = tin())`
`plot(dtm_tin)`

Redcozmo, 2023-10-02:
No it does not work like with this exemple : 

```
library(rgl)
with(iris, plot3d(Sepal.Length, Sepal.Width, Petal.Length, type="s", col=as.numeric(Species)))
```

I've just found this issue : [180](https://github.com/r-lidar/lidR/issues/180)

And for me it works when I run `rglwidget()` after `plot(las, size = 3, bg = "white")`

And a when loading rgl library an explicite message is printed

```
library('rgl')
This build of rgl does not include OpenGL functions.  Use
 rglwidget() to display results, e.g. via options(rgl.printRglwidget = TRUE).
```
And in my first code I did not have loading rgl library.

closed 2023-10-02 by Jean-Romain

Redcozmo, 2023-10-02:
Yes not lidR ! Thanks.
But I'm on Ubuntu