r-lidar/lidR #718: Dalponte2016 returns NA values

ouroukhai, 2023-10-10:
Hello and good morning,

I created the attached file by using the following code (fws ais mentioned in the sequel):

```r
opt_filter(nctg) <- "-keep_class 5 && -keep_z 0 50"
# set parallel
plan(multisession, workers = 4L)

# find tree tops by the max raster
ttops <- locate_trees(chm, 
                      lmf(ws = fws, shape = "circular"), 
                      uniqueness = "bitmerge")

# algorithm to apply
algo <- dalponte2016(chm, ttops)
opt_output_files(nctg) <- paste0(folder_path, "/dalponte2016_segm", "/{*}_dalponte2016_segm")
nctg_segm <- segment_trees(nctg, algo, 
                           uniqueness = "bitmerge")

# create lax
lidR:::catalog_laxindex(nctg_segm)
``` 
I have also attached the chm file that I used for the separation of tree tops. The fws is used as a draft function to reproduce the bug, as I have the same problem with other functions. It can be found in the following code block:
```r
fws <- function(x, intercept = 2) {
  y <- 2.6 * (-(exp(-0.08*(x-2)) - 1)) + intercept
  y[x < 2] <- intercept
  y[x > 20] <- 5
  return(y)
}
``` 
My problem is that the algorithm returns NA values of treeID on clearly high-vegetation trees. I saw no mention in the documentation of why the algorithm would return NA values; that's why it troubles me.
[files.zip](https://github.com/r-lidar/lidR/files/12853882/files.zip)
Note: The file is a clipped circle (150m radius, xcenter = 309300,  ycenter = 4322500) on the original file to avoid upload size issues. The chm is a 1kmx1km region containing the circle. Data come from NEON 2019 UKFS region. Please let me know if the original file is needed to add a link.
Thank you for your time!

ouroukhai, 2023-10-10:
It was a mistake on my end regarding the chosen resolution for creating the chm file. By increasing the pixel size, the number of NA values, in the treeID variable, was decreased.

Jean-Romain, 2023-10-10:
Your problem is resolved?

ouroukhai, 2023-10-10:
I still Observe some NA values but not in the same frequency as before. For reference, I work on ~9.5 points/m^2 and I resolve the problem by going up to 1m x 1m pixels. If the only reason that NA values are introduced is because of empty pixels, then I am okay. I guess the 9.5 points density is estimated and not exact, so having some empty pixels is expected? It is really strange though to see whole trees classified as NAs.

Jean-Romain, 2023-10-10:
If the CHM has NA i guess the map of the trees will have NAs at the same locations.

closed 2023-10-10 by ouroukhai