r-lidar/lidR #729: rasterize_terrain() only outputs the last tile of LAScatalog

natedailey7, 2023-11-16:
When I use `rasterize_terrain()` to create a DTM based on a LAScatalog, only the last tile is shown in my output. The same happens when I use `normalize_height()`. This is a continuation of a GIS Stack Exchange post [here](https://gis.stackexchange.com/questions/470222/normalizing-lascatalog-only-outputs-one-tile). 

I uploaded my data (2 las files) along with my script to Google Drive [here](https://drive.google.com/file/d/10qnsih1_8aeH0-JGkV_ToQ1tmLYMAKSP/view?usp=sharing).

```
install.packages("lidR")
library(lidR)

WORKING_DIRECTORY <- "G:/Shared drives/DCR Team Drive/Workspace/Tehama_Evacuation_Strategy/project_data/gray_pine_classification/11.15.23_reproducible_error"
setwd(WORKING_DIRECTORY)

# Assign relative paths for las files and R Outputs. 
LAS_CATALOG <- "data"
R_OUTPUT_FILES <- "R_Outputs"

# read the two las files
ctg <- readLAScatalog(LAS_CATALOG)

plot(ctg)
```
[image]

```
# Assign location for output files and allow overwriting. 
opt_output_files(ctg) <- R_OUTPUT_FILES
ctg@output_options$drivers$SpatRaster$param$overwrite <- TRUE

# create dtm. For me, only the last tile is shown in the plot. 
dtm <- rasterize_terrain(ctg, algorithm =knnidw(), overwrite=TRUE)
plot(dtm)
```
[image]
```
# Normalize height -- again, only the last tile is processed for me. 
norm <- normalize_height(ctg, tin())
plot(norm)
```
[image]

Let me know if you have any ideas or suggestions -- thank you so much.

Jean-Romain, 2023-11-16:
```r
R_OUTPUT_FILES <- "R_Outputs"
opt_output_files(ctg) <- R_OUTPUT_FILES
```

Your path is not templated. So it processes the first LAS file, writes it in `R_ouput.tif` then the second LAS file also in `R_output.tif` and erases the first raster. Use the name of the LAS files to name your rasters

```r
R_OUTPUT_FILES <- "{ORIGINALFILENAME}_chm"
opt_output_files(ctg) <- R_OUTPUT_FILES
```

I could have answered on stack exchange if you provided a reproducible code :wink: :

closed 2023-11-16 by Jean-Romain

natedailey7, 2023-11-16:
AMAZING!! Worked perfectly -- Thank you so much 😄