r-lidar/lidR #732: Unexpected behaviour classify_roi()

zoeschindler, 2023-11-21:
Hello, I have an issue with `classify_roi()`.
I want to classify points within some polygons (derived from buffered lines) as a different class than the remaining points.
But when I read my data in using `readTLSLAS()` instead of `readLAS()`, I get weird results. I made some dummy data: https://drive.google.com/file/d/1OOl4_vRYCIRSmx-Ey7cflgZv_s-y9l07/view?usp=sharing

Here is the code:

```
# load packages
library(lidR)
library(sf)

# read pointcloud
las <- readTLSLAS("test.las")
crs(las) <- "EPSG:32632"

# load shapes
shp_lines <- st_read("test.shp")
shp_polys <- st_buffer(shp_lines, 0.025)

# show pointcloud & shapes
plot(las)
plot(shp_polys)

# classify using lines
las$Classification <- LASUNCLASSIFIED
las <- classify_poi(las, class = LASGROUND, roi = shp_polys)

# show results
plot(las, color = "Classification")
plot(filter_poi(las, Classification == LASGROUND))
```

Polygons:
[image]

Result when using `readLAS()` (expected result):
[image]

Result when using `readTLSLAS()` (unexpected result):
[image]

Jean-Romain, 2023-11-21:
Thank you for the nice and easily reproducible example. I confirm the bug.

Jean-Romain, 2023-11-22:
Minimal Reproducible example

```r
las <- readLAS("~/Téléchargements/lidR issues/issue 732/test.las")
st_crs(las) <- "EPSG:32632"
shp_lines <- st_read("~/Téléchargements/lidR issues/issue 732/test.shp")
shp_polys <- st_buffer(shp_lines, 0.025)
sfc <- sf::st_geometry(shp_polys)
sum(lidR:::point_in_polygons(las, sfc), na.rm = T)
#> TLS: 101111
#> ALS: 469881
```

By the way @zoeschindler  do not use `crs()`. Use `st_crs()`. `crs()` will be removed in next version to avoid problem on CRAN with old geospatial package (`raster`, `sp`)

zoeschindler, 2023-11-22:
Thank you, will do :)

zoeschindler, 2023-11-22:
Sorry, pressed the wrong button xD Thanks for mentioning that. I will change my code to `st_crs()`.

Jean-Romain, 2023-11-22:
Fixed with a temporary fix. My octree has a problem, so I disabled the octree and used another spatial indexing structure by default.

commit 3bf2bf0b (2023-11-22): Fix #732