r-lidar/lidR #733: automatic merge of catalog_apply fails

spono, 2023-11-30:
Ciao JR,
I'm delineating the tree crowns from a large dataset running this:

```
ctg <- segment_trees(ctg, dalponte2016(chm, ttops)) 

crowns <- crown_metrics(ctg, func = .stdtreemetrics, geom = "convex")
```

and I got this error:

```
An error occurred when processing the chunk 22. Try to load this chunk with:
 chunk <- readRDS("C:\Users\...\AppData\Local\Temp\Rtmpy8SNGg/chunk22.rds")
 las <- readLAS(chunk)
no applicable method for 'st_agr<-' applied to an object of class "NULL"

Class attribute on column 5 of item 4 does not match with column 5 of item 1.Warning messages:
1: No tree found. NULL returned. 
2: An error occured during the automatic merge of 'catalog_apply'. Merging is impossible. A list has been returned.
```
your fail-safe code outputs a list of `sf` objects which get perfectly merged with `dplyr::bind_rows`.
I quickly checked the point clouds and I confirm that the tile that failed (i.e. `chunk22` mentioned above) didn't have any segmented object in it.

Is that the desired behaviour? Would be possible to fix it allowing to skip to the next tile or to remove NULL values before merging?

Dealing with high mountain areas it is not rare that a tile covers only high elevation rocky spots with nothing to segment.

Jean-Romain, 2023-12-07:
I'm not sure to understand. The code failed while processing chunk 22 or did it process everything and failed at the end while merging the partial outputs?

Jean-Romain, 2023-12-08:
drop null is supposed to be already true. Anyway, you are not supposed to used it. Do the following looks like a valid reproducible example ?

```r
library(lidR)
LASfile <- system.file("extdata", "MixedConifer.laz", package="lidR")
ctg = readLAScatalog(LASfile)
las <- readLAS(LASfile, select = "xyz", filter = "-inside 481250 3812980 481300 3813030")
ttops <- locate_trees(las, lmf(ws = 5))

res = list(ttops, list(NULL), ttops)
engine_merge(ctg, res)
```

closed 2023-12-11 by Jean-Romain

Jean-Romain, 2023-12-11:
Fixed. The case was originally handled but moving to `sf` reintroduced it.

commit ecf3066a (2023-12-11): Fix #733