r-lidar/lidR #735: LAS unreadable with QGIS after exported it without chang with lidR::write

Redcozmo, 2024-01-16:
Hi,

I encountered an issue with reading a LAS file exported with lidR in QGIS, without any change...
Issue or understanding problem from me ? LidR or QGis ?

1 - I read the LAS file
2 - I export with `writeLAS` function
3 - I try to read it with QGIS and I have this error : 

```
Invalid Data Source : /data/LHD_FXX_0843_6530_PTS_O_LAMB93_IGN69_nochange.copc.laz is 
not a valid or recognized data source, error: "Unable to recognize /data/LHD_FXX_0843_6530_PTS_O_LAMB93_IGN69_nochange.copc.laz as a LAZ file: """
```

I can read the original LAS file with QGIS.
I can read the exported LAS with lidR.

You can download the LAS here :
https://storage.sbg.cloud.ovh.net/v1/AUTH_63234f509d6048bca3c9fd7928720ca1/ppk-lidar/OL/LHD_FXX_0843_6530_PTS_O_LAMB93_IGN69.copc.laz

Code : 
```
# Set working directory
setwd(dirname(rstudioapi::getActiveDocumentContext()$path))

# Libraries
library(lidR) # to manipulate LiDAR data

# Read LAS
input_filename <- "LHD_FXX_0843_6530_PTS_O_LAMB93_IGN69.copc.laz"
output_filename <- "LHD_FXX_0843_6530_PTS_O_LAMB93_IGN69_nochange.copc.laz"
input_filepath <- paste(getwd(), input_filename, sep = "/")
output_filepath <- paste(getwd(), output_filename, sep = "/")
las <- readLAS(input_filepath)

# Write LAS
lidR::writeLAS(las = las, file = output_filepath)
```

Thanks a lot !!

Jean-Romain, 2024-01-16:
My best guess: lidR does not write copc files. It seems that QGIS is parsing the `.copc.laz` extension without checking if the file is really a copc file or not. Your file **is not** a copc file. Try to name it `.laz`

Redcozmo, 2024-01-16:
Yes it's that !! Thank you very much.

closed 2024-01-16 by Redcozmo

Jean-Romain, 2024-01-16:
[`lasR`](https://github.com/r-lidar/lasR) can write COPC files. Also my development version (still on my computer) of `rlas` can do it so my version of lidR can do it. I could upload it if you want to test.

Redcozmo, 2024-01-17:
I keep in mind `lasR` package that I didn't know but for the moment `lidR` is already a good and sufficient learning for me. Thanks !