r-lidar/lidR #736: Segmentation algorithm request

ouroukhai, 2024-01-24:
Hello lidR team,
and thank you for the great package you provide.

I wanted to make a request on a specific tree segmentation algorithm if possible. The algorithm can be found in this article [Wang et al 2017](https://ieeexplore.ieee.org/stamp/stamp.jsp?tp=&arnumber=7493651) under the code name _FGI-P_. It is a voxel based algorithm, and according to the study seems to outperforms many of the approaches discuss in the paper.

Thank you in advance for the consideration in the matter.
Best regards,

Jean-Romain, 2024-01-24:
Is there an open source code shared by the authors ?

ouroukhai, 2024-01-25:
Under my current understanding, no. In the article, there is no mention of the implementation code. I could send a personal message to the authors asking for an implementation code if this could assist further. I will mention the possibility of implementing this code for this package.

Jean-Romain, 2024-01-26:
Without access to an existing implementation, I won't do it. 

I did it in the past. It takes weeks of work and usually:

- I need to guess some parts that are not actually described enough in the papers to be reproducible
- I get extremely poor results at the end because everybody is claiming to have the best algorithm, but it is always tested in a super restricted dataset and compared to a watershed in the best case.
- The algorithm is so complex that in practice it can't scale beyond the size of a plot.

So now I'm no longer working without existing implementation. The balance between pain and gain is very biased toward pain.

But with an existing implementation as reference, I'm really keen to work on such a feature.

closed 2024-02-06 by Jean-Romain