r-lidar/lidR #749: `readLAS()` leads to RStudio crash when applied on a corrupt file

dimfalk, 2024-03-27:
When applying `readLAS()` on a corrupted file, {lidR} causes the IDE to crash at the moment:

``` r
library(lidR)
#> 4.1.1

f <- "3dm_32_299_5634_1_nw.laz"

utils::download.file(paste0("https://www.opengeodata.nrw.de/produkte/geobasis/hm/3dm_l_las/3dm_l_las/", f), 
                     f)

lidR::readLAS(f)
#> --- CRASH ---
```
[image]

I'm fully aware that this is due to the wrong `mode` in `utils::download.file()`, but I feel like a package should not cause a crash of the superordinate piece of software and also that it would have been nice to get a meaningful error message, in order to be able to investigate further instead of the upper notification.

``` r
f <- "3dm_32_299_5634_1_nw.laz"

utils::download.file(paste0("https://www.opengeodata.nrw.de/produkte/geobasis/hm/3dm_l_las/3dm_l_las/", f), 
                     f,
                     mode = "wb")

lidR::readLAS(f)
#> class        : LAS (v1.2 format 1)
#> memory       : 517.8 Mb 
#> extent       : 299000, 3e+05, 5634000, 5635000 (xmin, xmax, ymin, ymax)
#> coord. ref.  : ETRS89 / UTM zone 32N 
#> area         : 1 km²
#> points       : 8.48 million points
#> density      : 8.48 points/m²
#> density      : 8 pulses/m²
```

Just to show how other packages handle this: 
When messing things up using e.g. {terra}, i.e. GDAL, you get the following error message. Something similiar would be nice, I guess.

``` r
f <- "snw_NAM-44_CCCma-CanESM2_historical-r1_r1i1p1_CCCma-CanRCM4_r2_mon_195001-195012.nc"

download.file(url = paste0("https://crd-data-donnees-rdc.ec.gc.ca/CCCMA/products/CanSISE/output/CCCma/CanRCM4/NAM-44_CCCma-CanESM2_historical-r1/mon/atmos/snw/r1i1p1/", f),
              destfile = f)

terra::rast(f)
#> Warning:
#> `snw_NAM-44_CCCma-CanESM2_historical-r1_r1i1p1_CCCma-CanRCM4_r2_mon_195001-195012.nc'
#> not recognized as a supported file format. (GDAL error 4)
#> Error: [rast] cannot open this file as a SpatRaster: snw_NAM-44_CCCma-CanESM2_historical-r1_r1i1p1_CCCma-CanRCM4_r2_mon_195001-195012.nc
```

Jean-Romain, 2024-03-27:
Please report your operating system, `rlas` version, R version and everything that might be useful. I cannot reproduce on my side (linux, rlas 1.7.0, lidR 4.1.1, R 4.3.3). Your file is read properly.

dimfalk, 2024-03-27:
Sorry, I missed that. Here you go:

``` r
sessionInfo()
#> R version 4.3.2 (2023-10-31 ucrt)
#> Platform: x86_64-w64-mingw32/x64 (64-bit)
#> Running under: Windows 10 x64 (build 19045)
#> 
#> Matrix products: default
#> 
#> 
#> locale:
#> [1] LC_COLLATE=German_Germany.utf8  LC_CTYPE=German_Germany.utf8   
#> [3] LC_MONETARY=German_Germany.utf8 LC_NUMERIC=C                   
#> [5] LC_TIME=German_Germany.utf8    
#> 
#> time zone: Europe/Berlin
#> tzcode source: internal
#> 
#> attached base packages:
#> [1] stats     graphics  grDevices utils     datasets  methods   base     
#> 
#> other attached packages:
#> [1] rlas_1.7.0 lidR_4.1.1
#> 
#> loaded via a namespace (and not attached):
#>  [1] dplyr_1.1.4        compiler_4.3.2     tidyselect_1.2.0   reprex_2.1.0      
#>  [5] Rcpp_1.0.12        yaml_2.3.8         fastmap_1.1.1      R6_2.5.1          
#>  [9] generics_0.1.3     classInt_0.4-10    sf_1.0-15          knitr_1.45        
#> [13] tibble_3.2.1       units_0.8-5        R.cache_0.16.0     DBI_1.2.2         
#> [17] pillar_1.9.0       R.utils_2.12.3     rlang_1.1.3        utf8_1.2.4        
#> [21] terra_1.7-71       xfun_0.42          fs_1.6.3           cli_3.6.2         
#> [25] withr_3.0.0        magrittr_2.0.3     class_7.3-22       digest_0.6.34     
#> [29] grid_4.3.2         rstudioapi_0.15.0  lifecycle_1.0.4    R.methodsS3_1.8.2 
#> [33] R.oo_1.26.0        vctrs_0.6.5        KernSmooth_2.23-22 proxy_0.4-27      
#> [37] evaluate_0.23      glue_1.7.0         data.table_1.15.2  styler_1.10.2     
#> [41] codetools_0.2-19   fansi_1.0.6        e1071_1.7-14       rmarkdown_2.26    
#> [45] purrr_1.0.2        pkgconfig_2.0.3    tools_4.3.2        htmltools_0.5.7
```

Ok, maybe I should try updating R to 4.3.3 in the meantime.

Jean-Romain, 2024-03-27:
Give it a try. I currently have no clue at all, but it is more likely a Windows specific bug in `rlas`