r-lidar/lidR #759: R crashes using CSF on some clouds

EvanZarrella, 2024-05-28:
Running the `classify_ground` function with the CSF algorithm on some point clouds leads to R crashing. 
The CSF parameters : `lidR::csf(sloop_smooth = FALSE, iterations = 1000L, class_threshold = 0.5, cloth_resolution = 0.2)`
Tested on both R v4.3.2 and v4.4.0 running on an AMD Ryzen 9 7850X3D, 64Gb of RAM and a RTX 4080.
Using CloudCompare's implementation of the CSF gives satisfactory results. 

I've put on my personnal drive a file that's crashing.
https://drive.google.com/file/d/1s3eehS0lKVO-wFlreCCP6UNWJchPm1cg/view?usp=sharing

Let me know if I can be of further help investigating this issue or if I need to post this on the RCSF github issue tracker.

EvanZarrella, 2024-05-28:
Thank you for your response.

Regarding the OS, I'm on Windows 11, latest version.
Regarding more information, the `cloth_resolution` argument seems to be issue as plugin in 0.3 doesn't lead to any crashes 
Tried with your code, crashed on the classify_ground so it is a reproductible example for me. 
Tried the following code successfully :
```
las <- lidR::readLAS("plot_1000351_6769761.las")
data <- las@data
ground <- RCSF::CSF(data, sloop_smooth = FALSE, iterations = 1000L, class_threshold = 0.5, cloth_resolution = 0.2)
```
Leads me to believe that it comes from lidR and not the RCSF package.

Jean-Romain, 2024-05-28:
I confirm it crashes on Windows and I can identify the issue on Linux even if it does not crash. It comes from the code of the CSF. The code is not mine. I'm not sure I can fix it. Maybe I should check if the original CSF has been updated.

```
==73165== Invalid read of size 8
==73165==    at 0x1C2FB974: c2cdist::calCloud2CloudDist(Cloth&, csf::PointCloud&, std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&) (c2cdist.cpp:50)
==73165==    by 0x1C2EADE0: CSF::do_filtering(std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&, bool) (CSF.cpp:180)
==73165==    by 0x1C2F1588: R_CSF(Rcpp::DataFrame_Impl<Rcpp::PreserveStorage>, bool, double, double, int, int, double) (R_CSF.cpp:57)
==73165==    by 0x1C2F7BDE: _RCSF_R_CSF (RcppExports.cpp:21)
==73165==    by 0x495A237: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x495A88C: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49961FF: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B119C: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B150A: Rf_eval (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B36DE: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B44C6: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B4EEC: ??? (in /usr/lib/R/lib/libR.so)
==73165==  Address 0x1f423ae8 is 24 bytes after a block of size 32 in arena "client"
==73165== 
==73165== Invalid read of size 8
==73165==    at 0x1C2FB98B: c2cdist::calCloud2CloudDist(Cloth&, csf::PointCloud&, std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&) (c2cdist.cpp:51)
==73165==    by 0x1C2EADE0: CSF::do_filtering(std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&, bool) (CSF.cpp:180)
==73165==    by 0x1C2F1588: R_CSF(Rcpp::DataFrame_Impl<Rcpp::PreserveStorage>, bool, double, double, int, int, double) (R_CSF.cpp:57)
==73165==    by 0x1C2F7BDE: _RCSF_R_CSF (RcppExports.cpp:21)
==73165==    by 0x495A237: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x495A88C: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49961FF: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B119C: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B150A: Rf_eval (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B36DE: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B44C6: ??? (in /usr/lib/R/lib/libR.so)
==73165==    by 0x49B4EEC: ??? (in /usr/lib/R/lib/libR.so)
==73165==  Address 0x1f423bc0 is 32 bytes before a block of size 16 in arena "client"
==73165== 
```

We have pretty much the same without lidR

```r
las <- rlas::read.las("~/Téléchargements/plot_1000351_6769761.las")
ground <- RCSF::CSF(data, sloop_smooth = FALSE, iterations = 1000L, class_threshold = 0.5, cloth_resolution = 0.2)
```

```
==73621== Invalid read of size 8
==73621==    at 0x1C2FB974: c2cdist::calCloud2CloudDist(Cloth&, csf::PointCloud&, std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&) (c2cdist.cpp:50)
==73621==    by 0x1C2EADE0: CSF::do_filtering(std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&, bool) (CSF.cpp:180)
==73621==    by 0x1C2F1588: R_CSF(Rcpp::DataFrame_Impl<Rcpp::PreserveStorage>, bool, double, double, int, int, double) (R_CSF.cpp:57)
==73621==    by 0x1C2F7BDE: _RCSF_R_CSF (RcppExports.cpp:21)
==73621==    by 0x495A237: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x495A88C: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49961FF: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B119C: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B150A: Rf_eval (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B36DE: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B44C6: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B163B: Rf_eval (in /usr/lib/R/lib/libR.so)
==73621==  Address 0x1d01fae8 is 8 bytes before a block of size 32 free'd
==73621==    at 0x484C91D: operator delete(void*, unsigned long) (vg_replace_malloc.c:1101)
==73621==    by 0x1C2EF658: deallocate (new_allocator.h:145)
==73621==    by 0x1C2EF658: deallocate (alloc_traits.h:496)
==73621==    by 0x1C2EF658: _M_deallocate (stl_vector.h:354)
==73621==    by 0x1C2EF658: void std::vector<Particle*, std::allocator<Particle*> >::_M_realloc_insert<Particle* const&>(__gnu_cxx::__normal_iterator<Particle**, std::vector<Particle*, std::allocator<Particle*> > >, Particle* const&) (vector.tcc:500)
==73621==    by 0x1C2ECEA8: push_back (stl_vector.h:1198)
==73621==    by 0x1C2ECEA8: makeConstraint (Cloth.h:101)
==73621==    by 0x1C2ECEA8: Cloth::Cloth(Vec3 const&, int, int, double, double, double, double, int, double) (Cloth.cpp:75)
==73621==    by 0x1C2EAC44: CSF::do_filtering(std::vector<int, std::allocator<int> >&, std::vector<int, std::allocator<int> >&, bool) (CSF.cpp:141)
==73621==    by 0x1C2F1588: R_CSF(Rcpp::DataFrame_Impl<Rcpp::PreserveStorage>, bool, double, double, int, int, double) (R_CSF.cpp:57)
==73621==    by 0x1C2F7BDE: _RCSF_R_CSF (RcppExports.cpp:21)
==73621==    by 0x495A237: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x495A88C: ??? (in /usr/lib/R/lib/libR.so)
[... 49 lines cut ...]
==73621==    by 0x495A88C: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49961FF: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B119C: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B150A: Rf_eval (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B36DE: ??? (in /usr/lib/R/lib/libR.so)
==73621==  Block was alloc'd at
[... cut ...]
==73621==    by 0x1C2F1588: R_CSF(Rcpp::DataFrame_Impl<Rcpp::PreserveStorage>, bool, double, double, int, int, double) (R_CSF.cpp:57)
==73621==    by 0x1C2F7BDE: _RCSF_R_CSF (RcppExports.cpp:21)
==73621==    by 0x495A237: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x495A88C: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49961FF: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B119C: ??? (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B150A: Rf_eval (in /usr/lib/R/lib/libR.so)
==73621==    by 0x49B36DE: ??? (in /usr/lib/R/lib/libR.so)
==73621== 
```