r-lidar/lidR #760: Error in watershed algoritm

aitorhazi, 2024-05-30:
Hi, 

when I use the watershed algorithm to segment the trees it reports the following error:

Error in validImage(x) : object must be an array

I have converted the object into an array and when I run the watershed algorithm again it reports the following error:

Error: Invalid function provided as algorithm.

My code is:

nlas <- readLAS("C:/Users/5075-47595.laz",  filter = "-set_withheld_flag 0")
las <- normalize_height(nlas, knnidw())
col <- height.colors(50)
chm <- rasterize_canopy(las = las, res = 0.5, algorithm = p2r(0.15))
kernel <- matrix(1, 3, 3)
schm <- terra::focal(x = chm, w = kernel, fun = median, na.rm = TRUE)
algo = watershed(chm, th = 4)
las  = segment_trees(las, algo)

I have the EBImage package installed and the R version is 4.4.0

How can I overcome this obstacle?

Thanks.

Jean-Romain, 2024-05-30:
I ran this without issue. Please report a minimal reproducible example including data if you can't reproduce with embedded datasets

```r
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
nlas = readLAS(LASfile)
las <- normalize_height(nlas, knnidw())
col <- height.colors(50)
chm <- rasterize_canopy(las = las, res = 0.5, algorithm = p2r(0.15))
kernel <- matrix(1, 3, 3)
schm <- terra::focal(x = chm, w = kernel, fun = median, na.rm = TRUE)
algo <-  watershed(schm, th = 4)
SEGtrees  <-  segment_trees(las, algo)
```

aitorhazi, 2024-05-31:
Hi, 
I am attaching a screenshot with the workflow where the error appears.
I am also attaching below a link where you can download the .laz file used.
https://we.tl/t-D0iILYRaJN
Thank you so much.
[image]

Jean-Romain, 2024-06-07:
The link expired

aitorhazi, 2024-06-10:
I attach it again. 

https://we.tl/t-GwKzZzidKK

The link expires in a week.

Thanks.

Jean-Romain, 2024-06-10:
Ok I get the files thanks.

Jean-Romain, 2024-06-10:
I ran

```r
library(lidR)

nlas <- readLAS("~/Téléchargements/issue 760/5075-47595.laz",  filter = "-set_withheld_flag 0")
las <- normalize_height(nlas, knnidw())
col <- height.colors(50)
chm <- rasterize_canopy(las = las, res = 0.5, algorithm = p2r(0.15))
kernel <- matrix(1, 3, 3)
schm <- terra::focal(x = chm, w = kernel, fun = median, na.rm = TRUE)
algo <-  watershed(schm, th = 4)
SEGtrees  <-  segment_trees(las, algo)
```

I worked. Please report more information.

closed 2024-06-22 by Jean-Romain