r-lidar/lidR #762: Non-API calls

Jean-Romain, 2024-06-22:
New CRAN policy check

```r
Compiled code should not call non-API entry points in R.
See ‘Writing portable packages’ in the ‘Writing R Extensions’ manual.
```

Easy:

* `R_curErrorBuf`

Problematic:

* `SETLENGTH` to make resizable vectors

We are not alone with this problem:

* https://github.com/Rdatatable/data.table/issues/6180
* https://github.com/r-lib/vctrs/issues/1933

We may need to remove the `point_metrics()` function if we can't find a solution

Jean-Romain, 2024-07-04:
Removed `point_metrics()` temporarily

samuelgrubinger, 2025-06-08:
Hi JR,
Is there any plan to reinstate point_metrics()? 
It would be very helpful. If it were available in lidR, I wouldn't need to do such 'fancy' things in lasR. 
Unfortunately, I am working on a cloud computing system and can only install from conda-forge, so can't grab it from the devel version.

Jean-Romain, 2025-06-09:
You can install it from r-universe. It's not my fault if CRAN changes its policies without providing a fallback. This also affects, for example, the data.table package, and they haven’t found a workaround yet. If world-class programmers can’t find a solution, I probably won’t either. Still, data.table remains on CRAN, maybe they can put more pressure on CRAN, I don’t know. If I release lidR with the feature, I’m pretty sure it will be kicked off CRAN.