r-lidar/lidR #764: Large area point cloud disappears when plotting with color="Z" and legend=TRUE

1iny, 2024-07-01:
Hi,

I've encountered an issue when plotting large area point clouds using lidR. The point cloud disappears when I set color = "Z" and legend = TRUE for areas around 100 hectares in size, if I set legend = FALSE, then no issue 
[image]
. This issue only occurs with relatively large area point clouds (even with low point density). I tried in different PCs and the same issue still exist.

Environment:
- R version: 4.3.3
- lidR version: 4.1.2
- Operating System: Windows 10
- RAM: 128GB

Code Example:
plot(large_cloud_las, color = "Z", legend = TRUE)

[image]

Jean-Romain, 2024-07-02:
How many points is large enough to reproduce?

1iny, 2024-07-02:
> How many points is large enough to reproduce?

My las file is about 200 points/sqm, area is 76 ha. I suspect its due to the area is too large. Cheers

Jean-Romain, 2024-07-02:
76 ha is 0.76 km² so 3/4 of the standard 1km² file. However you have approx 150 million points. `rgl` definitively can't handle that.

closed 2024-07-04 by Jean-Romain