r-lidar/lidR #769: opt_chunk_size

THLAB35, 2024-07-12:
Hello,

I encounter a problem during a step in the .las data processing process; can you help me? Sincerely

 opt_chunk_size(M24015_SYNERGIS_ETU_ENV_las) <- 500
Error in `opt_chunk_size<-`(`*tmp*`, value = 500) : 
  no slot of name "chunk_options" for this object of class "LAS"

Thibault

Jean-Romain, 2024-07-12:
`opt_chunk_size(ctg) = 500`

THLAB35, 2024-07-15:
Thank you; It's my mistake

I encounter the following problem:
Do you have any possible solutions?

> tin <- rasterize_terrain(las_A, res = 1, algorithm = tin())
Processing [==>-------------------------------------------------------------------]   4% (1/27) eta: 10mError: [writeRaster] file exists. You can use 'overwrite=TRUE' to overwrite it

Jean-Romain, 2024-07-15:
See https://gis.stackexchange.com/questions/463618/overwriting-files-in-function-pixel-metrics

closed 2024-07-16 by Jean-Romain