r-lidar/lidR #784: Calculation of pulse, point density off?

Silviculturalist, 2024-10-09:
Is it expected that the calculation of the pulse, point density / m**2 does not align with the manual calculation? 

```
library(lidR)
LASfile = system.file("extdata","example.laz",package='rlas')
las = readLAS(LASfile)
las = retrieve_pulses(las)
summary(las)

my_metrics <- function(pulseID){
return(
list(
nPulses = length(unique(pulseID))
)
)
}

result = cloud_metrics(las,~my_metrics(pulseID))
result$nPulses/area(las)
```

summary(las) provides pulse density as 2.63 pulses/m2 whereas manual calculation gives 2.83?

Jean-Romain, 2024-10-09:
The summary looks only at pre-computed number of first return while `retrieve_pulse` used the gpstime attribute in order to compute the exact pulse ID of each point. This way if you have a 2nd or 3rd return only without the first one (e.g. on the edge of the dataset).

The difference should be minimal in most cases but here you chose an extraordinary tiny dataset (30 points) where a single point difference can change a lot the final value.

Silviculturalist, 2024-10-17:
Yes, difference is minimal, noticed some decimal differences on my real dataset. 
Small dataset here only as reprex. 
Maybe it can be clarified in your documentation for summary function ? 

Something like - # pulses, first return only?

closed 2025-02-04 by Jean-Romain