r-lidar/lidR #786: Cannot install lidR on R 3.6.3 due to cpp 14+ requirements

quinn-r88, 2024-11-07:
# Problem

Attempting to install lidR on R 3.6.3 yields the following errors:
```
In file included from concaveman/concaveman.h:24,
                 from convaveman.cpp:2:
concaveman/rtree.h: In member function ‘void rtree<T, DIM, MAX_CHILDREN, DATA>::insert(rtree<T, DIM, MAX_CHILDREN, DATA>::data_type, const bounds_type&)’:
concaveman/rtree.h:44:21: error: ‘make_unique’ is not a member of ‘std’
   44 |       auto r = std::make_unique<type>(data, bounds);
      |                     ^~~~~~~~~~~
concaveman/rtree.h:44:21: note: ‘std::make_unique’ is only available from C++14 onwards
concaveman/rtree.h:44:37: error: expected primary-expression before ‘>’ token
   44 |       auto r = std::make_unique<type>(data, bounds);
      |                                     ^
concaveman/rtree.h:67:22: error: ‘make_unique’ is not a member of ‘std’
   67 |     auto leaf = std::make_unique<type>(best_child.get().data(), best_child.get().bounds());
      |                      ^~~~~~~~~~~
concaveman/rtree.h:67:22: note: ‘std::make_unique’ is only available from C++14 onwards
concaveman/rtree.h:67:38: error: expected primary-expression before ‘>’ token
   67 |     auto leaf = std::make_unique<type>(best_child.get().data(), best_child.get().bounds());
      |                                      ^
make: *** [/usr/lib/R/etc/Makeconf:177: convaveman.o] Error 1
ERROR: compilation failed for package ‘lidR’
* removing ‘/home/qbowers/R/x86_64-pc-linux-gnu-library/3.6/lidR’
```

# Steps to Reproduce

1. Install lidR dependencies per the README
    - `sudo add-apt-repository ppa:ubuntugis/ubuntugis-unstable`
    - `sudo apt-get update`
    - `sudo apt-get install libgdal-dev libgeos++-dev libudunits2-dev libproj-dev libx11-dev libgl1-mesa-dev libglu1-mesa-dev libfreetype6-dev libxt-dev libfftw3-dev`
2. Install the `terra` package from R-Universe (the CRAN version similarly does not install on R 3.6.3)
    - `install.packages('terra', repos='https://rspatial.r-universe.dev')`
3. Install `lidR`
    - `packages.install('lidR')`

# My Setup

- OS: Ubuntu 20.04
- R: 3.6.3

# Commentary

I know that 3.6.3 is reasonably out of date, however the lidR CRAN package still claims to support R versions as old as 3.5. Ideally a user should not have to look up the history of R's default compiler settings to install a package if it claims to be supported. Would it be possible to either change the CRAN package to report that it only supports R 4+, or else add back the C++ version requirements somewhere, perhaps as described [here](https://www.tidyverse.org/blog/2023/03/cran-checks-compiled-code/)?

Thanks!

Jean-Romain, 2024-11-08:
The issue isn’t with R 3.6 itself but with the fact that it compiles by default with C++11, whereas lidR requires C++14. Previously, this was enforced by explicitly adding the C++14 flag in the Makevars file. However, with the transition to R 4.0, then 4.1, and subsequent versions, CRAN has progressively moved to default C++ standards—first C++11, then C++14, and now C++17. CRAN specifically requested that I remove the explicit C++14 flag from Makevars: https://github.com/r-lidar/lidR/commit/072d031b64295e76fcd2ff5a2f47ae384cd0e262. As a result, without this explicit C++14 specification, it compiles under C++11 with R 3.6. While R 3.6 does support C++14, the recent CRAN policy changes mean that this information is no longer enforced in older R versions.

quinn-r88, 2024-11-08:
Yeah, I did see those spec changes. I notice though that other CRAN packages are still able to compile with C++14 or higher, even on old R versions. [This article](https://www.tidyverse.org/blog/2023/03/cran-checks-compiled-code/) describes using a configure script to the same effect, which CRAN seems to prefer - why they would prefer that over just stating the c++ version I cannot say, but at least it works.

Here's one example of such a script: https://github.com/tidyverse/readxl/pull/722

Or alternatively, you could just change the description to require R 4.0+ or something

Jean-Romain, 2024-11-08:
Thank you for sharing the link. Can you make a PR ?

quinn-r88, 2024-11-12:
Done! Check out #787 when you've got a chance.

PR #787 (2024-12-17): Use autoconf to generate Makevars at install time
This PR adds a new `configure` script to the package, which will generate Makevar files when the package is installed. The `configure` script is generated by GNU autoconf from `configure.ac`, and must be regenerated by the developer when `configure.ac` is modified.

This PR also adds a `cleanup` script, which R will use to remove the files that `configure` will generate.

Resolves #786 

## References:
- https://www.tidyverse.org/blog/2023/03/cran-checks-compiled-code/
- https://github.com/tidyverse/readxl/pull/722
- https://cran.r-project.org/ [... cut]
closed 2024-12-17 by Jean-Romain