r-lidar/lidR #793: Improve `las_check()` CRS checking

ppoyk, 2024-12-19:
Function `las_check()` reports "Inconsistent CRS across files" when checking a `LAScatalog` object with multiple LAS file formats.

In my case, after separately checking each file included in the catalog, all files did have identical CRS information. 
This contradicts with the output of `las_check()`.

I suspect this has to do with only some files in the catalog having a WKT defined. LAS files with < 1.4 specification lack WKT information.
(Checking WKTs is supposedly included in the "CRS check" section of `las_check()`.)

Possible solution would be to only check file CRSs, output the check results, and warn the user if also WKTs are present.
Another solution would be to convert WKTs / CRSs to each other. Then check that all files concur. Warning should also be raised if any file has both CRS and WKT defined, but they do not equal after conversions.

Jean-Romain, 2024-12-19:
In my opinion this is the expected behavior. las_check is giving you a warning that there is something to check. You checked it. It is ok. End of the story.

ppoyk, 2024-12-20:
You are correct that the general chek results should guide the user to make more detailed investigations.

Jean-Romain, 2024-12-20:
My concern with that suggestion is that it gonna be very tedious for a small gain. CRS interpretation is prone to minor variation. For example in you case are you sure that the EPSG code returns the **exact** same CRS as the WKT string?

closed 2025-02-04 by Jean-Romain

ppoyk, 2025-02-05:
I am not that familiar with the conversion process.
There is potential to check the equality of the stored WKT to the WKT represented by the EPSG code using e.g. https://spatialreference.org/.
This issue may remain closed.