r-lidar/lidR #795: `st_transform` raises `Error in geom[[1]][, 3] : subscript out of bounds`

Kodiologist, 2025-01-08:
```R
library(lidR)
library(sf)

x = readLAS("USGS_LPC_RI_Statewide_D22_350000_260000.laz")
  # From https://rockyweb.usgs.gov/vdelivery/Datasets/Staged/Elevation/LPC/Projects/RI_Statewide_D22/RI_Statewide_1_D22/LAZ/USGS_LPC_RI_Statewide_D22_350000_260000.laz
# Warning: There are 3965 points flagged 'withheld'.
st_transform(x, "epsg:6567")
# Error in geom[[1]][, 3] : subscript out of bounds
```

Traceback:

```
6: st_transform.LAS(x, "epsg:6567") at transform.R#86
5: st_transform(x, "epsg:6567") at lidr_bug_repro.R#8
4: eval(ei, envir)
3: eval(ei, envir)
2: withVisible(eval(ei, envir))
1: source("lidr_bug_repro.R")
```

I'm using R 4.4.0, `lidR` 4.1.2, `sf` 1.0.19, GEOS 3.8.0, GDAL 3.0.4, and PROJ 6.3.1 on Ubuntu 20.04.

Kodiologist, 2025-01-08:
I think this might be related to using an old version of PROJ. The example works fine on another machine with newer versions of various dependencies, and on the problematic machine, `st_transform` also exhibits various failures when trying to transform other objects (such as `terra` rasters or `sf` simple feature objects) to EPSG:6567.

Jean-Romain, 2025-01-08:
Ok so I guess I can close this issue ?

Kodiologist, 2025-01-08:
I don't know. `lidR` may need to be more robust to documented cases in which the underlying reprojection logic fails, or there may be a bug that was fixed in later versions of the dependencies. I don't really know what's going on.

closed 2025-01-08 by Jean-Romain