r-lidar/lidR #796: Extracting other canopy metrics than max height with rasterize_canopy()

jonathanvonoppen, 2025-01-22:
Hi Jean-Romain,

first, thank you for developing and maintaining this package that makes working with LiDAR data nice and accessible!

For `rasterize_canopy()`, have you considered enabling extraction of different canopy height metrics than maximum height per cell? I am working in an alpine area with low vegetation (mostly sub-metre height), and leaving aside the inherent uncertainty in Z value, `median` height, for instance, might be more representative of the actual structure of the vegetation than maxima that could, for instance, represent single flowering structures during peak growing season. 

I tried modifying the `p2r()` function accordingly as 
``` {r} 
p2r_median = function(subcircle = 0, na.fill = NULL)
{
  assert_is_a_number(subcircle)
  assert_all_are_non_negative(subcircle)
  
  subcircle <- lazyeval::uq(subcircle)
  na.fill   <- lazyeval::uq(na.fill)
  
  if (!is.null(na.fill) && !is(na.fill, "SpatialInterpolation"))
    stop("'na.fill' is not an algorithm for spatial interpolation")
  
  f = function(las, layout)
  {
    assert_is_valid_context(LIDRCONTEXTDSM, "p2r")
    
    dsm <- fasterize::fasterize(las, layout, subcircle, "median")  # changed fun = "max" to "median"
    
    if (!is.null(na.fill))
    {
      verbose("Interpolating empty cells...")
      
      layout[[1]][] <- NA_real_
      layout[[1]][] <- dsm
      hull <- st_convex_hull(las)
      hull <- sf::st_buffer(hull, dist = raster_res(layout)[1])
      
      where <- raster_as_dataframe(layout, xy = FALSE, na.rm = FALSE)
      where <- where[is.na(where$Z)]
      where <- sf::st_multipoint(as.matrix(where[,1:2]))
      where <- sf::st_geometry(where)
      where <- sf::st_set_crs(where, sf::st_crs(hull))
      where <- sf::st_intersection(where, hull)
      where <- sf::st_coordinates(where)
      if (ncol(where) == 3L) where = where[,-3L]
      where <- as.data.frame(where)
      data.table::setnames(where, c("X", "Y"))
      
      grid  <- raster_as_las(layout)
      
      lidR.context <- "p2r"
      cells <- raster_cell_from_xy(layout, where$X, where$Y)
      layout[[1]][cells] <- na.fill(grid, where)
      
      dsm <- as.numeric(layout[[1]])
    }
    
    return(dsm)
  }
  
  f <- plugin_dsm(f)
  return(f)
}
```
, however, 
``` {r}
library(lidR)
LASfile <- system.file("extdata", "MixedConifer.laz", package ="lidR")
las <- readLAS(LASfile)
chm <- rasterize_canopy(las, res = 1, algorithm = p2r_median())
```
returns
`Error in UseMethod("wk_handle") : 
  no applicable method for 'wk_handle' applied to an object of class "LAS"`

From its documentation, `fasterize::fasterize()` only 'currently takes "sum", "first", "last", "min", "max", "count", or "any"'. As far as I understand it, that is because point clouds within a cell are being transformed to convex hulls before extracting the maximum Z value. I tried with `terra::rasterize()` instead, but of course that cannot handle LAS objects.

Do you see any chance of implementing the extraction of canopy metrics beyond maximum height?

Many thanks,
Jonathan

jonathanvonoppen, 2025-01-22:
Nevermind, seems like `pixel_metrics()` is exactly what I need. Thanks for that!

closed 2025-01-22 by jonathanvonoppen

Jean-Romain, 2025-01-22:
> Nevermind, seems like pixel_metrics() is exactly what I need. Thanks for that!

Absolutely

For your code the problem likely comes from

```r
dsm <- fasterize::fasterize(las, layout, subcircle, "median")  # changed fun = "max" to "median")
```

The `fasterize::fasterize´ function signature is 

```r
fasterize(
  sf,
  raster,
  field = NULL,
  fun = "last",
  background = NA_real_,
  by = NULL
)
```