r-lidar/lidR #798: Error when using rasterize_terrain on LAScatalog saved with previous versions ?

jmmonnet, 2025-01-30:
Hi Jean-Romain,

I downloaded thousands of LAZ files from the french national dataset a few months ago and then saved the corresponding catalog in a .rda file. I used `rasterize_terrain` to produce tif files from selected files without issues, but I recently got an error.

For example, the following code throws an error when trying to rasterize one of the files:
```
# file to process
las_file <- "/media/reseau/drones/Lidar/LidarHD/data/LAZ_classe/PM/LHD_FXX_0913_6455_PTS_O_LAMB93_IGN69.copc.laz"
# load previously saved catalog
load(file = "/media/reseau/drones/Lidar/LidarHD/data/LAZ_classe/cata.rda")
# keep only one file of interest (for comparison with next code)
cata@data <- cata@data[basename(cata$filename) == basename(las_file),]
# set output files
lidR::opt_output_files(cata) <- paste0("/media/reseau/drones/Lidar/LidarHD/data/temp/test/dtm10m_{ORIGINALFILENAME}")
future::plan(future::multisession, workers = 2)
dtm <- lidR::rasterize_terrain(cata, 10, lidR::tin())
```
The error is `Erreur : aucun slot de nom "ptr" pour cet objet de la classe "SpatRaster"`

The code runs without error if I build a new catalog with this same file.
```
cata <- lidR::catalog(las_file)
lidR::opt_output_files(cata) <- paste0("/media/reseau/drones/Lidar/LidarHD/data/temp/test/dtm10m_{ORIGINALFILENAME}")
future::plan(future::multisession, workers = 2)
dtm <- lidR::rasterize_terrain(cata, 10, lidR::tin())
```
I guess the error is due to some updates in lidR or terra, it took me quite a while to figure out that it was linked to the catalog being saved with an older version. Maybe issuing a warning to the user would help debugging errors when processing catalogs saved with previous versions... I can try to produce a reprex if that helps.

Thank you !
JM

jmmonnet, 2025-02-04:
I am sorry I cannot produce an example with data from the package as the error appears when processing a catalog that was saved previously and I do not have such archives. 
Here is a [link to download other example files](https://filesender.renater.fr/?s=download&token=cf98930e-4112-4780-a606-b19e18c9a321). The catalog was saved in February 2024. I guess packages were up to date.
Then if I run the following code, I get an error when using rasterize_terrain on the catalog loaded from the saved file, but it runs ok if I use a new one. (lidR 4.1.2 and terra 1.8-15)
The error is different from the one I got previously.

```
# load previously saved catalog
load(file = "cata.rda")
# keep only one file (the one available for download)
cata@data <- cata@data[4,]
# replace absolute filename with relative one
cata$filename <- basename(cata$filename)
lidR::opt_filter(cata) <- "-keep_every_nth 10"
lidR::opt_output_files(cata) <- paste0("{ORIGINALFILENAME}")
dtm <- lidR::rasterize_terrain(cata, 10, lidR::tin())
# Erreur : aucun slot de nom "cpp" pour cet objet de la classe "SpatRaster"

#
# new catalog with same file
cata_new <- lidR::catalog(cata$filename)
# save it
save(cata_new, file = "cata_new.rda")
# remove object
rm(cata_new)
# load it
load("cata_new.rda")
lidR::opt_filter(cata_new) <- "-keep_every_nth 10"
lidR::opt_output_files(cata_new) <- paste0("{ORIGINALFILENAME}")
dtm <- lidR::rasterize_terrain(cata_new, 10, lidR::tin())
# no error
```

closed 2025-02-04 by Jean-Romain

jmmonnet, 2025-02-04:
Thank you for the fast answer, I expected some backward compliance issues related to changes in packages. Thank you also for the `vpc` tip, it seems a convenient alternative.

Jean-Romain, 2025-02-04:
Please report any issue with lidR and VPC. It has not been thoroughly tested in lidR.