r-lidar/lidR #799: Chunk processing using LAScatalog leaving behind gaps on some chunk boundaries

jamestisdale, 2025-02-04:
I am having an issue with LAScatalog output having missing gaps after chunk processing. I am using a chunk size of 50 m^2. and after processing using a function that filters the point cloud to returns that are near the ground (i.e., 0.1 - 1.3 m), Here is the function: 

process_las <- function(las) {
  tryCatch({
    if (is(las, "LAScatalog")) {
      opt_select(las) <- "*" # disable select tuning
      options <- list(
        automerge = TRUE,
        need_output_file = TRUE,
        need_buffer = TRUE # requirements
      )
      output <- catalog_apply(las, process_las, .options = options)
      return(output)
    } else if (is(las, "LAScluster")) {
      # Read in the LAS cluster
      x <- readLAS(las)
      if (npoints(x) == 0) return(NULL)
      x@header@PHB$x.offset <- x_offset
      x@header@PHB$y.offset <- y_offset
      x@header@PHB$z.offset <- z_offset
      glas <- classify_ground(x, mycsf) # CSF parameters from above
      nlas <- normalize_height(glas, tin()) # using tin() algorithm
      flas <- filter_poi(nlas, Z >= 0.1, Z <= 1.3) # Meters
      filt_csf <- classify_noise(flas, sor(k = 10, m = 0)) # noise filter
      filt_csf <- filter_poi(filt_csf, Classification != 18) # delete noise
      last_filter <- filter_poi(filt_csf, buffer == 0) # removes buffer 
      return(last_filter)
    } else if (is(las, "LAS")) {
      x <- readLAS(las)
      x@header@PHB$x.offset <- x_offset
      x@header@PHB$y.offset <- y_offset
      x@header@PHB$z.offset <- z_offset
      glas <- classify_ground(x, mycsf) # CSF parameters from above
      nlas <- normalize_height(glas, tin()) # using tin() algorithm
      flas <- filter_poi(nlas, Z >= 0.1, Z <= 1.3) # Meters
      filt_csf <- classify_noise(flas, sor(k = 10, m = 0)) # noise filter
      filt_csf <- filter_poi(filt_csf, Classification != 18) # delete noise
      last_filter <- filter_poi(filt_csf, buffer == 0)
      return(last_filter)
    } else {
      stop("Not supported input")
    }
  }, error = function(e) {
    # Log the error message and skip processing this chunk
    message(paste("Error processing chunk:", e$message))
    return(NULL) # Return NULL to indicate failure but continue processing
  })
}

Here is an image of the output after processing. Does it process like this if nothing is found in those areas? 

[image]

Jean-Romain, 2025-02-04:
You are keeping the points that are in [0.1, 1.3], then you apply a very aggressive sor and discard noise. You write the few remaining points and you are chunking in very very small section. Without seeing the dataset I can only make guess and it looks like a plausible outcome.

Jean-Romain, 2025-02-04:
Simplify your code for better maintainability

```r
process_las <- function(las) {
  tryCatch({
    if (is(las, "LAScatalog")) 
    {
      opt_select(las) <- "*" # disable select tuning
      options <- list(
        automerge = TRUE,
        need_output_file = TRUE,
        need_buffer = TRUE # requirements
      )
      output <- catalog_apply(las, process_las, .options = options)
      return(output)
    } 
   else if (is(las, "LAScluster"))
   {
      # Read in the LAS cluster
      x <- readLAS(las)
      if (npoints(x) == 0) return(NULL)
      return(process_las(x))
    } 
    else if (is(las, "LAS")) 
    {
      glas <- classify_ground(las, mycsf) # CSF parameters from above
      nlas <- normalize_height(glas, tin()) # using tin() algorithm
      flas <- filter_poi(nlas, Z >= 0.1, Z <= 1.3) # Meters
      filt_csf <- classify_noise(flas, sor(k = 10, m = 0)) # noise filter
      filt_csf <- filter_poi(filt_csf, Classification != 18) # delete noise
      last_filter <- filter_poi(filt_csf, buffer == 0)
      return(last_filter)
    } 
   else {
      stop("Not supported input")
    }
  }, error = function(e) {
    # Log the error message and skip processing this chunk
    message(paste("Error processing chunk:", e$message))
    return(NULL) # Return NULL to indicate failure but continue processing
  })
}
```

jamestisdale, 2025-02-04:
Okay. Thank you for your input on my function! 

So LAS_apply will trim the chunks if no points are in the square area?

Jean-Romain, 2025-02-04:
> So LAS_apply will trim the chunks if no points are in the square area?

No, it just produce a valid LAS file in which the header records the bounding box. The bounding box is defined by the points recorded. If you do not have a lot of points I can reduce the effective bounding box compare to your expectation.

jamestisdale, 2025-02-04:
Okay, thank you for the help!