r-lidar/lidR #800: `catalog()` used on a VPC saved from `lasR::write_vpc(, absolute_path = T)` does not build path correctly

jmmonnet, 2025-02-04:
Hi,

It seems that the file path is not built correctly when a LAScatalog is read from a `vpc` file written by `lasR::write_vpc()` with option `absolute_path = T`. The file path is ok when using relative `absolute_path = F`.

Please find an example below.
JM

```
# build catalog from file
cata <- lidR::catalog(system.file("extdata", "Topography.laz", package="lidR"))
# create output directory
dir.create("./vpc")
#
# save as vpc, absolute path
vpc_file <- "./vpc/cata.vpc"
pipeline = lasR::write_vpc(vpc_file, absolute_path = T)
lasR::exec(pipeline, on = cata)
# read catalog from vpc
cata_from_vpc <- lidR::catalog(vpc_file)
# Message d'avis :
# Dans normalizePath(absolute_path) :
#   path[1]="/home/jean-matthieu_monnet/R/lidar/lidarHD/vpc//usr/lib/R/site-library/lidR/extdata/Topography.laz": Aucun fichier ou dossier de ce nom
# filename
cata$filename
cata_from_vpc$filename
#
# save as vpc, relative path
pipeline = lasR::write_vpc(vpc_file, absolute_path = F)
lasR::exec(pipeline, on = cata)
# read catalog from vpc
cata_from_vpc <- lidR::catalog(vpc_file)
# filename
cata$filename
cata_from_vpc$filename
```

Jean-Romain, 2025-02-06:
Fixed

closed 2025-02-06 by Jean-Romain

Jean-Romain, 2025-09-02:
If the issue is different than https://github.com/r-lidar/lasR/issues/205 please open a new issue with a MRE. i.e no extra function and ensuring the issue is not related to https://github.com/r-lidar/lasR/issues/205

commit 0a09bcb8 (2025-02-06): Fix #800 VPC files