r-lidar/lidR #804: Using self-defined pixel_metrics() function in LAScatalog

alandhyc, 2025-04-12:
A few of my colleagues have spent a lot of time trying to solve these issues, so I though I might bring it up here. Thanks!

(1) 
The first issue is that ```terra``` masks ```is.empty()``` from lidR, so if terra is loaded, ```catalog_map()``` fails with the error ```Error: unable to find an inherited method for function 'is.empty' for signature 'x = "LAS"'```. The solution is simple, don't load ```terra``` separately, but I wonder whether someone could change the ```if (is.empty(las)) return(NULL)``` line to ```if (lidR::is.empty(las)) return(NULL)``` in ```catalog_map()``` so this issue doesn't occur in the future? I spent a lot of time figuring out what caused this and I suspect it'll happen to someone else in the future.

(2) 
pixel_metrics() doesn't seem to be able to find objects in the environment when used in catalog_map() or catalog_apply(). This issue seems to occur for both self-defined functions (q99_f) or other variables (z_threshold). This makes it quite problematic when you have to calculate multiple metrics over and over again. I seem to recall some related issues being brought up in the past, but I gather that it hasn't been resolved yet?

Thank you!

```
library(lidR)
library(future)
library(terra) #1st bug, this induces: "Error: unable to find an inherited method for function 'is.empty' for signature 'x = "LAS"'"

metrics_exp_f<-function(las,res,z_threshold){

  require(terra)
    
  if(is(las,"LAS")){
  
  #Define a function
    
  q99_f<-function(z,zt){
    z<-z[which(z>zt)]
    list(p_99 = quantile(z,
                         probs = 0.99))
  }
  
  #Use function to calculate metrics
  
  q99<-pixel_metrics(las,
                     res = res,
                     func = ~q99_f(z = Z,        #2nd bug, "Error: could not find function "q99_f""
                                   zt = z_threshold))  #3rd bug, similar to 2nd, couldn't find object"
  
  return(q99)
  
  }
  
  if(is(las,"LAScatalog")){
    
    options<-list(
      need_output_file=T
    )
    
      mtrc_result<-catalog_map(las,
                               metrics_exp_f,
                               res = res,
                               z_threshold = z_threshold,
                               .options = options)
    
  } #End of is(las,"LAScatalog")
  
}

#Apply function

ncores<-2
plan(multisession, workers = ncores)
set_lidr_threads(ncores)

ctg<-readLAScatalog("E:/LiDAR/temp_las_low_pass")

out_dir<-"E:/LiDAR/LiDAR_metrics"

opt_output_files(ctg)<-paste0(out_dir,"/{ORIGINALFILENAME}")
opt_wall_to_wall(ctg)<-F
opt_merge(ctg)<-F

metrics_exp_f(las = ctg,
              res = 5,
              z_threshold = 1.3)

```

Jean-Romain, 2025-04-12:
First issue is know. `terra` added `is.empty` years after `lidR` and created a clash. This should be a relatively recent addition in `terra` because I discovered the issue only recently.

Second issue is not an issue here I think. Your code is overly complex and contains recursive calls, undefined functions, nested functions. If I understand well, all you need is:

```r
library(lidR)

q99_f<-function(z,zt)
{
  z<-z[z>zt]
  list(p_99 = quantile(z, probs = 0.99))
}

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg = readLAScatalog(LASfile)
pixel_metrics(ctg, ~q99_f(Z, 1.3), 5)
```

alandhyc, 2025-04-13:
Thanks Jean-Romain! 

(1) To clarify, I did not call ```is.empty()```, it is hidden inside ```catalog_map()```, which makes it very difficult to debug. I was just wondering whether someone could change to source code of ```catalog_map()``` to ensure that it doesn't happen again. 

(2) Maybe I should better explain myself. I'm trying to write a wrapper function ```lidar_metrics``` that calculates many self-defined indices at once based on different subsets of the point cloud (based on [this paper](https://www.sciencedirect.com/science/article/abs/pii/S0034425722004321)). The example provided is highly simplified (the full function being hundreds of lines long). It would be nice if pixel_metrics() work inside a function so one doesn't have to repeat hundreds of lines of code over and over again.

Jean-Romain, 2025-04-13:
(1) Ho I see. This is a serious issue. Please open an issue with a minimal reproducible example

(2) I see. `pixel_metrics()`  is not able to find object outside the underlying `dataframe`. This does not work because `zt` cannot be found. This is a long term issue known for 10 years know

```r
custom_wrapper = function(las, zt, res = 5)
{
  pixel_metrics(las, ~q99_f(Z, zt), res)
}
```
It needs to be

```r
custom_wrapper = function(las, res = 5)
{
  pixel_metrics(las, ~q99_f(Z, 1.3), res)
}
```

Yet, `pixel_metrics` works in lidR. So there is a possible trick. You need to capture and substitute in the expression.

```r
custom_wrapper = function(las, zt, res = 5)
{
  expr <- substitute(q99_f(Z, zt_val), list(zt_val = zt))
  pixel_metrics(las, expr, res)
}
```

However this won't work because `q99_f is not visible in the global environment

```r
custom_wrapper = function(las, zt, res = 5)
{
  q99p_f<-function(z,zt)
  {
    z<-z[z>zt]
    list(p_99 = quantile(z, probs = 0.99))
  }
  
  expr <- substitute(q99p_f(Z, zt_val), list(zt_val = zt))
  pixel_metrics(las, expr, res)
}
```

If you function is not visible in the global environnement a option is to make it visible through a package

```r
custom_wrapper = function(las, zt, res = 5)
{
  expr <- substitute(pkg::q99_f(Z, zt_val), list(zt_val = zt))
  pixel_metrics(las, expr, res)
}
```
If  `custom_wrapper` is in `pkg` `q99_f` should be visible as well.

Jean-Romain, 2025-04-15:
What about

```r
custom_wrapper = function(las, zt = 1, res = 5)
{
  expr <- substitute(~pkg::q99_f(Z, zt_val), list(zt_val = zt))
  pixel_metrics(las, eval(expr), res)
}
``

alandhyc, 2025-04-16:
That makes a lot of sense, it works now, package up and running! Thanks so much!

Jean-Romain, 2025-04-17:
Benchmark it against a classical call. Just to check there is no big overhead.

closed 2025-04-27 by Jean-Romain