r-lidar/lidR #805: Issue with catalog_map() caused by conflict between terra and lidR

alandhyc, 2025-04-15:
This issue was discussed in #804, but it was suggested that I create a new thread. Essentially ```catalog_map()``` calls ```is.empty()``` under the hood, which causes problems if the function is masked by ```terra::is.empty()```. As the user did not call ```is.empty()```, it's often not obvious why the error occurs. 

Here is an example: 

```
library(lidR)
library(future)

#Very simple function that creates 

stretch<-function(las){
  
  require(terra)
  
  if(is(las,"LAS")){
    
    las_stretched<-las
    las_stretched$Z<-las_stretched$Z*3

    rast<-rasterize_canopy(las_stretched, res = 1, algorithm = p2r())
    
    # las<-pixel_metrics(las,~mean(Z),res = 5)
    
    return(rast)
    
  }
  
  if(is(las,"LAScatalog")){
    
    options<-list(
      need_output_file=T
    )
    
    result<-catalog_map(las,
                        stretch,
                        .options = options)
    
  } #End of is(las,"LAScatalog")
  
}

#Apply function to catalog

ncores<-2
plan(multisession, workers = ncores)
set_lidr_threads(ncores)

ctg<-readLAScatalog("E:/Aland/LiDAR/temp_las_low_pass")

out_dir<-"E:/Aland/LiDAR/test"
opt_output_files(ctg)<-paste0(out_dir,"/{ORIGINALFILENAME}")

stretch(las = ctg)
```

.las files used in the example:
[plot_7_283473.9_801978.4.zip](https://github.com/user-attachments/files/19759944/plot_7_283473.9_801978.4.zip)

Thank you!

Jean-Romain, 2025-05-16:
There is no problem in this code. You use bad patrice by including a package in a function. I don't know what kind of side effect is effectively happening here but with correct code there is no issue

```r
library(lidR)
library(future)
library(terra)

#Very simple function that creates 
stretch<-function(las)
{
  if(is(las,"LAS"))
  {
    las_stretched<-las
    las_stretched$Z<-las_stretched$Z*3
    rast<-rasterize_canopy(las_stretched, res = 1, algorithm = p2r())
    return(rast)
  }
  
  if(is(las,"LAScatalog"))
  {
    
    options<-list(need_output_file=T)
    result<-catalog_map(las,stretch,.options = options)
  }
}

#Apply function to catalog

ncores<-2
plan(multisession, workers = ncores)
set_lidr_threads(ncores)

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
ctg<-readLAScatalog(LASfile)

out_dir<-tempdir()
opt_output_files(ctg)<-paste0(out_dir,"/{ORIGINALFILENAME}")

stretch(las = ctg)
``

closed 2025-05-16 by Jean-Romain

Jean-Romain, 2025-05-17:
Actually I think this was already fixed in a previous commit. It will be available on CRAN soon.