r-lidar/lidR #807: Error with point classification greater than 31

jamie-carter, 2025-04-30:
I have a topobathy lidar data set that includes classes higher than 31. One class in particular, '40', represents bathymetric points. I'd like to include this class in a rasterize_terrain function but readLAS fails to load the .laz file. Instead, it throws the following errors.

_ERROR: '' needs arguments between 0 and 31 but '40' is out of range
Error: Filter error see message above._

Note that this also fails when I try to classify_terrain or rasterize_terrain using the same .laz files loaded into a LAScatalog.

Jean-Romain, 2025-04-30:
try `"-keep_extended_class 2 40"`

jamie-carter, 2025-05-01:
> try `"-keep_extended_class 2 40"`

Thank you, that works for the readLAS function. Class 40 is used when running rasterize_terrain on the LAS object (even without specifying the use_class argument).

_las_2 <- readLAS(files = c("laz/2022_391500e_4822000n_las.laz", "laz/2022_391500e_4822500n_las.laz"), filter = "-keep_class 2")
las_2_rast <- rasterize_terrain(las_2)

las_2_40 <- readLAS(files = c("laz/2022_391500e_4822000n_las.laz", "laz/2022_391500e_4822500n_las.laz"), filter = "-keep_extended_class 2 40")
las_2_40_rast <- rasterize_terrain(las_2_40)

dtms <- c(las_2_rast, terra::crop(las_2_40_rast, las_2_rast))
names(dtms) <- c("Class 2", "Classes 2 and 40")
plot(dtms)_

[image]

However, the behavior changes when working with LAScatalogs.

_ctg_2 <- readLAScatalog("laz/", filter = "-keep_class 2")
ctg_2_40 <- readLAScatalog("laz/", filter = "-keep_extended_class 2 40")
ctg_2_rast <- rasterize_terrain(ctg_2)
ctg_2_40_rast <- rasterize_terrain(ctg_2_40)

ctgs <- c(ctg_2_rast, terra::crop(ctg_2_40_rast, ctg_2_rast))
names(ctgs) <- c("Catalog: Class 2", "Catalog: Classes 2 and 40")
plot(ctgs)_

[image]

When I try using the use_class argument in rasterize_terrain, the function fails when processing the first tile.

_ctg_2_40_rast_useClass <- rasterize_terrain(ctg_2_40, use_class = c(2L, 40L))_

**Processing [=========================================>------------------------------------------]  50% (1/2) eta:  0sError: Filter error see message above.**

Jean-Romain, 2025-05-01:
What is the error?

jamie-carter, 2025-05-01:
It produces a very short error message:

_Error: Filter error see message above._

Jean-Romain, 2025-05-01:
Try to produce a minimal reproducible example

closed 2025-05-16 by Jean-Romain