r-lidar/lidR #810: wrong COPC extent

wiesehahn, 2025-05-27:
Hey,
there seems to be some problem with Cloud Optimized Point Clouds.
After converting a las file to copc with lasR it seems to be read errorneous by lidR. 

`lasR::info()` reports the same extent for .laz and .copc.laz, the same is true when reading the header information. 

But reading the data with lidR reports invalid data and and a wrong extent. 

``` r
library(lasR)

infile <- system.file("extdata", "Megaplot.las", package="lasR")
outfile <- tempfile(fileext = ".copc.laz")

exec(
    write_las(ofile = outfile),
  on = infile
)
#> [1] "C:\\Users\\JWIESE~1\\AppData\\Local\\Temp\\Rtmpe8cDt7\\fileed7c75cc27a0.copc.laz"

info(infile)
#> Source       : LASF (v1.2)
#> Size         : 2.53 MB
#> Extent       : 684766.39 684993.29 5017773.08 5018007.25 (xmin, xmax, ymin, ymax)
#> Points       : 81.59 thousands
#> Density      : 1.5 pts/mÂ²
#> Coord. ref.  : NAD83 / UTM zone 17N
#> Schema       :
#> 17 attributes | 31 bytes per points
#>  Name: flags             | uchar  | Desc: Internal 8-bit mask reserved for lasR core engine
#>  Name: X                 | int    | Desc: X coordinate
#>  Name: Y                 | int    | Desc: Y coordinate
#>  Name: Z                 | int    | Desc: Z coordinate
#>  Name: Intensity         | ushort | Desc: Pulse return magnitude
#>  Name: ReturnNumber      | uchar  | Desc: Pulse return number for a given output pulse
#>  Name: NumberOfReturns   | uchar  | Desc: Total number of returns for a given pulse
#>  Name: Classification    | uchar  | Desc: The 'class' attributes of a point
#>  Name: UserData          | uchar  | Desc: Used at the userâ€™s discretion
#>  Name: PointSourceID     | short  | Desc: Source from which this point originated
#>  Name: ScanAngle         | char   | Desc: Rounded angle at which the laser point was output
#>  Name: gpstime           | double | Desc: Time tag value at which the point was observed
#>  Name: EdgeOfFlightline  | bit    | Desc: Set when the point is at the end of a scan
#>  Name: ScanDirectionFlag | bit    | Desc: Direction in which the scanner mirror was traveling 
#>  Name: Synthetic         | bit    | Desc: Point created by a technique other than direct observation
#>  Name: Keypoint          | bit    | Desc: Point is considered to be a model key-point
#>  Name: Withheld          | bit    | Desc: Point is supposed to be deleted)

info(outfile)
#> Source       : LASF (v1.4)
#> Size         : 2.86 MB
#> Extent       : 684766.39 684993.29 5017773.08 5018007.25 (xmin, xmax, ymin, ymax)
#> Points       : 81.59 thousands
#> Density      : 1.5 pts/mÂ²
#> Coord. ref.  : NAD83 / UTM zone 17N
#> Schema       :
#> 19 attributes | 35 bytes per points
#>  Name: flags             | uchar  | Desc: Internal 8-bit mask reserved for lasR core engine
#>  Name: X                 | int    | Desc: X coordinate
#>  Name: Y                 | int    | Desc: Y coordinate
#>  Name: Z                 | int    | Desc: Z coordinate
#>  Name: Intensity         | ushort | Desc: Pulse return magnitude
#>  Name: ReturnNumber      | uchar  | Desc: Pulse return number for a given output pulse
#>  Name: NumberOfReturns   | uchar  | Desc: Total number of returns for a given pulse
#>  Name: Classification    | uchar  | Desc: The 'class' attributes of a point
#>  Name: UserData          | uchar  | Desc: Used at the userâ€™s discretion
#>  Name: PointSourceID     | short  | Desc: Source from which this point originated
#>  Name: ScanAngle         | float  | Desc: Angle at which the laser point was output
#>  Name: ScannerChannel    | uchar  | Desc: Channel (scanner head) of a multi-channel system
#>  Name: gpstime           | double | Desc: Time tag value at which the point was observed
#>  Name: EdgeOfFlightline  | bit    | Desc: Set when the point is at the end of a scan
#>  Name: ScanDirectionFlag | bit    | Desc: Direction in which the scanner mirror was traveling 
#>  Name: Synthetic         | bit    | Desc: Point created by a technique other than direct observation
#>  Name: Keypoint          | bit    | Desc: Point is considered to be a model key-point
#>  Name: Withheld          | bit    | Desc: Point is supposed to be deleted)
#>  Name: Overlap           | bit    | Desc: If set, point is within an overlap region of 2+ swaths

lidR::readLASheader(infile)
#> File signature:           LASF 
#> File source ID:           0 
#> Global encoding:
#>  - GPS Time Type: GPS Week Time 
#>  - Synthetic Return Numbers: no 
#>  - Well Know Text: CRS is GeoTIFF 
#>  - Aggregate Model: false 
#> Project ID - GUID:        00000000-0000-0000-0000-000000000000 
#> Version:                  1.2
#> System identifier:        LAStools (c) by rapidlasso GmbH 
#> Generating software:      las2las (version 171231) 
#> File creation d/y:        0/0
#> header size:              227 
#> Offset to point data:     321 
#> Num. var. length record:  1 
#> Point data format:        1 
#> Point data record length: 28 
#> Num. of point records:    81590 
#> Num. of points by return: 55756 21493 3999 342 0 
#> Scale factor X Y Z:       0.01 0.01 0.01 
#> Offset X Y Z:             0 0 0 
#> min X Y Z:                684766.4 5017773 0 
#> max X Y Z:                684993.3 5018007 29.97 
#> Variable Length Records (VLR):
#>    Variable Length Record 1 of 1 
#>        Description: by LAStools of rapidlasso GmbH 
#>        Tags:
#>           Key 1024 value 1 
#>           Key 3072 value 26917 
#>           Key 3076 value 9001 
#>           Key 4099 value 9001 
#> Extended Variable Length Records (EVLR):  void

lidR::readLASheader(outfile)
#> File signature:           LASF 
#> File source ID:           0 
#> Global encoding:
#>  - GPS Time Type: GPS Week Time 
#>  - Synthetic Return Numbers: no 
#>  - Well Know Text: CRS is WKT 
#>  - Aggregate Model: false 
#> Project ID - GUID:        00000000-0000-0000-0000-000000000000 
#> Version:                  1.4
#> System identifier:         
#> Generating software:       
#> File creation d/y:        0/0
#> header size:              375 
#> Offset to point data:     1643 
#> Num. var. length record:  1 
#> Point data format:        6 
#> Point data record length: 30 
#> Num. of point records:    81590 
#> Num. of points by return: 55756 21493 3999 342 0 0 0 0 0 0 0 0 0 0 0 
#> Scale factor X Y Z:       0.01 0.01 0.01 
#> Offset X Y Z:             0 0 0 
#> min X Y Z:                684766.4 5017773 0 
#> max X Y Z:                684993.3 5018007 29.97 
#> Variable Length Records (VLR):
#>    Variable Length Record 1 of 1 
#>        Description: by LAStools of rapidlasso GmbH 
#>        WKT OGC COORDINATE SYSTEM: PROJCRS["NAD83 / UTM zone 17N",BASEGEOGCRS["NAD83",DATUM["North Americ [...] (truncated)
#> Extended Variable Length Records (EVLR):  void
```

``` r
inlas <- lidR::readLAS(infile)

print(inlas)
#> class        : LAS (v1.2 format 1)
#> memory       : 4.4 Mb 
#> extent       : 684766.4, 684993.3, 5017773, 5018007 (xmin, xmax, ymin, ymax)
#> coord. ref.  : NAD83 / UTM zone 17N 
#> area         : 51572 m²
#> points       : 81.6 thousand points
#> density      : 1.58 points/m²
#> density      : 1.08 pulses/m²

outlas <- lidR::readLAS(outfile)
#> Warning: Invalid data: 1 points with a return number equal to 0 found.
#> Warning: Invalid data: 1 points with a number of returns equal to 0 found.

print(outlas)
#> class        : LAS (v1.4 format 6)
#> memory       : 5 Mb 
#> extent       : 0, 684993.3, 0, 5018007 (xmin, xmax, ymin, ymax)
#> coord. ref.  : NAD83 / UTM zone 17N 
#> area         : 648 km²
#> points       : 81.6 thousand points
#> density      : 0 points/m²
#> density      : 0 pulses/m²
```
<sup>Created on 2025-05-27 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-05-27:
This looks indeed very bad....

Jean-Romain, 2025-05-27:
That was a surprisingly easy fix and a very good timing because I planned to track a bug in COPC writer today. Thank to your bug report you gave me a MRE that saved me a lot of time. However I still have trouble since the produced file look COPC valid but not read as COPC.

Jean-Romain, 2025-05-27:
Juste a quick question. Where is it documented that `write_las()` supports COPC in `lasR`. As far a I can tell it is not documented.

wiesehahn, 2025-05-27:
> Just a quick question. Where is it documented that `write_las()` supports COPC in `lasR`. As far a I can tell it is not documented.

its not officially documented, you mentioned it here: https://github.com/r-lidar/lasR/discussions/101#discussioncomment-11494197

Jean-Romain, 2025-05-27:
# lasR 0.14.0

- Doc: Documented that lasR can write COPC files.
- New: Added write_copc() function with extra arguments to control hierarchy depth and density.
- Fix: Fixed unmapped memory issue when writing COPC files.

If there are other problems please open a dedicated issue

Jean-Romain, 2025-05-27:
Why is this issue in `lidR`? Moving