r-lidar/lidR #820: `rasterize_canopy` raises "C++ unexpected internal error in 'filter_with_grid': point out of raster"

Kodiologist, 2025-08-01:
Here's an odd one. It appears to be very sensitive to the input, so it's probably a strange numerical issue.

```R
library(lidR)

filename = "USGS_LPC_CT_Statewide_C16_155895_sw.laz"
if (!file.exists(filename))
    download.file(
        paste0("https://rockyweb.usgs.gov/vdelivery/Datasets/Staged/Elevation/LPC/Projects/CT_Statewide_C16/CT_Statewide_B8_2016/LAZ/", filename),
        filename)

lidar = readLAS(
    filename,
    select = "xyzr",
    filter = "-drop_withheld -drop_synthetic -keep_xy 1156415 896070 1156480 896141")

f = \(res) rasterize_canopy(
    lidar,
    res = res,
    algorithm = pitfree())

f(3.2808333333333331)
  # Okay
f(3.2808333333333337)
  # Error: C++ unexpected internal error in 'filter_with_grid': point out of raster.
```

Jean-Romain, 2025-08-01:
Are you able to reproduce with normal resolution? Nobody is asking for a resolution of `3.2808333333333337`. So indeed we may have numerical issues. If the bug occurs with a regular resolution like `3.2` I'll investigate

Kodiologist, 2025-08-01:
> Are you able to reproduce with normal resolution?

No idea. All the different parts of this example seem interdependent, so it could be a difficult multi-parameter search problem to find another example that has a resolution with a smaller number of decimal places.

Kodiologist, 2025-08-01:
> Nobody is asking for a resolution of `3.2808333333333337`.

I did. That's how I found the crash. It's just the factor to convert from the US survey foot to meters, according to the units package.

Jean-Romain, 2025-08-01:
Ok but then you are requesting for a subatomic precision in meter units. Round your value. You will have troubles with terra as well if you have a full accuracy double precision value as a resolution.

Kodiologist, 2025-08-01:
I've definitely found some floating-point-related bugs in terra over the years (e.g., https://github.com/rspatial/terra/issues/709, https://github.com/rspatial/terra/issues/509), but they've gotten fixed. I'll report any more if I find them.

Kodiologist, 2025-08-01:
You mean this one? https://github.com/rspatial/terra/issues/844 Yeah, I think it's probably true that this example can't work quite as you would hope due to what 64-bit floating point can fit, in which case it's not really a bug.