r-lidar/lidR #824: readLAScatalog does not work with absolute file paths in VPC

wiesehahn, 2025-09-02:
As already mentioned in previous issues in https://github.com/r-lidar/lidR/issues/800#issuecomment-3244487839 and https://github.com/r-lidar/lidR/issues/771 readLAScatalog does not work with absolute file paths in Virtual Point Clouds.

``` r
packageVersion("lidR")
#> [1] '4.2.2'
packageVersion("lasR")
#> [1] '0.17.1'

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")

# since lasR currently does not produce absolute paths we edit this manually
vpc_rel <- lasR::exec(
  lasR::write_vpc(tempfile(fileext = ".vpc")), 
  on = LASfile)

t <- yyjsonr::read_json_file(vpc_rel)
t$features$assets[[1]]$data$href <- LASfile

vpc_abs <- tempfile(fileext = ".vpc")
yyjsonr::write_json_file(t, vpc_abs, pretty = TRUE)

ctg <- lidR::readLAScatalog(vpc_rel)
ctg <- lidR::readLAScatalog(vpc_abs)
#> Warning in normalizePath(path.expand(path), winslash, mustWork):
#> path[1]="C:/Users/jwiesehahn/AppData/Local/Temp/RtmpusNWqT/C:/Users/jwiesehahn/AppData/Local/Programs/R/R-4.5.0/library/lidR/extdata/Megaplot.laz":
#> Die Syntax für den Dateinamen, Verzeichnisnamen oder die Datenträgerbezeichnung
#> ist falsch
```

<sup>Created on 2025-09-02 with [reprex v2.1.1](https://reprex.tidyverse.org)</sup>

Jean-Romain, 2025-09-02:
This code works on my machine. Should be windows . Show me the output of

```r
packageVersion("lidR")
packageVersion("lasR")

LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")

# since lasR currently does not produce absolute paths we edit this manually
vpc_rel <- lasR::exec(
  lasR::write_vpc(tempfile(fileext = ".vpc")), 
  on = LASfile)

t <- yyjsonr::read_json_file(vpc_rel)
t$features$assets[[1]]$data$href

lidR:::is_absolute(t$features$assets[[1]]$data$href)

t$features$assets[[1]]$data$href <- LASfile

vpc_abs <- tempfile(fileext = ".vpc")
yyjsonr::write_json_file(t, vpc_abs, pretty = TRUE)

lidR:::read_vpc(vpc_rel)
lidR:::read_vpc(vpc_abs)
```

Jean-Romain, 2025-09-02:
What about

```r
lidR:::is_absolute(LASfile)
``

Jean-Romain, 2025-09-02:
show me `LASfile

wiesehahn, 2025-09-02:
`C:/Users/jwiesehahn/AppData/Local/Programs/R/R-4.5.0/library/lidR/extdata/Megaplot.laz`

Jean-Romain, 2025-09-02:
Ok I got it. My function `is_absolute` was assuming windows native separator `\` but R often use cross-platform `/` for windows.

commit 87de63eb (2025-09-02): Fix #824
closed 2025-09-02 by Jean-Romain